HASwitchPlate/HASPone #27: Would you like to check the 'hasp_Display_Alarm_Control_page7.yaml'

AlbertDijksma, 2021-03-07:
Hey,
Would you like to check the 'hasp_Display_Alarm_Control_page7.yaml'.
I get an error '2021-03-07 20:46:37 WARNING (SyncWorker_14) [homeassistant.components.manual.alarm_control_panel] Invalid code given for disarmed'
This problem is most likely caused in lines 364 and 389.
(states ("number.plate01_alarmcode_"

aderusha, 2021-03-07:
Can you explain a bit about what you're seeing in lines [364](https://github.com/HASwitchPlate/HASPone/blob/main/Home_Assistant/blueprints/hasp_Display_Alarm_Control_page7.yaml#L364) and [389](https://github.com/HASwitchPlate/HASPone/blob/main/Home_Assistant/blueprints/hasp_Display_Alarm_Control_page7.yaml#L389) that should be changed?

AlbertDijksma, 2021-03-07:
I think ".plate01" is a hardcoded name.

aderusha, 2021-03-07:
> I think ".plate01" is a hardcoded name.

I'll be damned - that's exactly what it is.  Thanks for the catch!  I'll fix this up and push a new one here in just a moment, then @ you here when that happens.  Thanks Albert!

AlbertDijksma, 2021-03-07:
Thanks for the quick response

aderusha, 2021-03-07:
Allright @AlbertDijksma I think I have this resolved.  What an embarrassing bug :D

Can you give [the new version](https://github.com/HASwitchPlate/HASPone/blob/main/Home_Assistant/blueprints/hasp_Display_Alarm_Control_page7.yaml) a shot and let me know how it goes for you?

AlbertDijksma, 2021-03-07:
It is working now!
Thanks

aderusha, 2021-03-07:
Thanks for the help on the bug report!

closed 2021-03-07 by aderusha