HASwitchPlate/HASPone #28: HASPone Overlay Issues

IIAIronWolf, 2021-03-08:
Reference discussion: #20 
Including what is mentioned in the above discussion, I have also noticed the page buttons stay white (as if they are active, unlike their normal grey, inactive state). This happens every now and then and required pressing different page buttons a few times until things go back to normal.
Can provide more info if necessary.

aderusha, 2021-03-08:
Can you explain the sequence of events you see?  The OP in the previous thread was dealing with the activate page blueprint, which you said you aren’t using.  What happens in your case?  What actions are you taking, and what happens in response?

IIAIronWolf, 2021-03-08:
> I am actually seeing the exact same issue as the OP on occasion, with the sticking icons, sometimes being the temperature icon. I am not using a activate page blueprint. I simply added the four blueprints you made in order to recreate the status page you had before HASPone, including date, time, temp, and weather condition. Therefore, either those blueprints have this same issue, or the Core Functionality blueprint may also need to be updated.
> In addition, and as you mentioned in this first comment, the page buttons sometimes stick in the 'activated' color, but those were only setup by core functionality on my device. I have not customized page buttons; therefore I believe this issue exists in the core functionality as well.

aderusha, 2021-03-08:
OK trying to get the exact sequence of events straight, so bear with me.

* You are starting on a page, I’ll call it page 1
* One page 1 is one or more blueprints which make use of the icon overlay
* You change pages using the center button of the three buttons on the bottom (b[2]),  which was set to go to page 2.
* Page 2 appears, blueprints assigned to this page appear, but with one or more icons from page 1 laid on top.

How am I doing?  Sound about right?

IIAIronWolf, 2021-03-08:
That is correct. That does happen sometimes. 
However, more often I am seeing this:

- Start on page one
- Click to page two
- Page one button (b[1]) is still highlighted white as if it is the active page.
- Click between pages until only one button is correctly showing highlighted as the active page.

aderusha, 2021-03-08:
Allright, sounds like two separate issues going on here so I don't want to conflate the two.  I'll start with the icons.  Any chance you can [grab a debug log](https://github.com/HASwitchPlate/HASPone/wiki/Capture-HASPone-debug-output) of the problem happening?

IIAIronWolf, 2021-03-09:
I rarely see the icon overlay issue anymore with HASPone; however I may be able to get your a debug log of the button active overlay issue.

aderusha, 2021-03-29:
Any updates on this issue?  Are you still experiencing the problem?

IIAIronWolf, 2021-04-28:
I've wiped out everything as of now. Sorry for the delayed response. For now, with my better understanding of how this works, I don't believe I am seeing the issue. I will reopen it if it occurs again.

closed 2021-04-28 by IIAIronWolf