HASwitchPlate/HASPone #30: HA with 6pcs HASPOne is very very slowly

proasnet, 2021-03-08:
Hi, today I configured 6pcs of HASPOne, each witch 24 automations.
Now is WebUI of HA very very slowly. If is core slowly for running automations in house, I dont know, I will know after Sunrise.
And a GUI of each HASPone is very slowly, I see it with page scrolling and re-drawing texts.
Before this update, all hasps was quick :(
Is it normaly?

aderusha, 2021-03-08:
Are all automations running slowly?  Or just HASP-related automations?  The rPi is a good starting point for Home Assistant but the performance is pretty low, particularly in regards to storage.  What rPi hardware are you using (pi3? or if 4, which model?), and what are you using for storage on that?

proasnet, 2021-03-09:
I am using RPI3 B+
After one day I find, that only HASP automations working slowly. Reaction of touch on any Hasp is a long, and listing of pages is too long.
As storage is only SD card in RPI with HA image 64bit, not any other storage.

aderusha, 2021-03-09:
[You might find this discussion helpful](https://old.reddit.com/r/homeassistant/comments/jvwtv1/friendly_reminder_dont_use_a_sd_card_on_a_pi/).  HASP fundamentally relies on Home Assistant automations, so if Home Assistant is running slow then HASP is going to be slow.  The linked thread is concerned with SD failures over their lifetime, but the same tips for minimizing writes to avoid killing your SD card also apply to increasing performance overall.  Disabling recorder and logger may make an impact for you.

The final suggestion in that post is what I'd generally recommend: buy a old Dell desktop off of eBay for not much money and Home Assistant will run like a champ.

proasnet, 2021-03-09:
I tried running HA on Synology NAS, before addon exist. Now, they have an Addon HA. I had HA as virtual machine, working good. Only one problem was a energy power of NAS 24/7, about 50 Watts with 5 HDDs.
Now I am powering NAS only if I need access to my files.
CPU performance of NAS was very high, so energy power too :)

aderusha, 2021-04-02:
Any update on this?  Has the suggestion above made any impact in performance?

proasnet, 2021-04-04:
Hello. Thank you for interesting. Performance is still very low. I tried
run HA on Synoligy NAS, that has better power than RPI but with no effect
to HASP speed. Problem is only with listing between pages. Reaction to call
scenes is instant and quick. There is no problem. Problem is with quick
loading of the items on any other pages. My wifi network is working perfect
with microtik devices and I have multiple APs in my house. Fixed channels
1, 6, 11 and diferent SSIDs.

aderusha, 2021-04-04:
Allright, so I think our next step is to figure out where things are moving slowly.  Can you [post a debug log](https://github.com/HASwitchPlate/HASPone/wiki/Capture-HASPone-debug-output) of page change process?  Looking for the whole conversation, from page select button press to the page being completely loaded.  If that looks good, then the next step will be to capture the MQTT traffic with timestamps.

proasnet, 2021-04-06:
Okay. As soon as possible I will write here logs from telnet and MQTT. Now, I am searching how I can get mqtt capture with timestamps from eclipse mosquitto, that run on my HA. I am still dont know how. I have only basic log in mqtt server.

aderusha, 2021-04-06:
If you have the mosquitto command line clients installed (`sudo apt-get install mosquitto-clients`), the following command will allow you to view messages with some debug info including timestamps.  Change the first few lines to match your environment.

```bash
MQTT_BROKER=hassio.local
MQTT_USER=hassio
MQTT_PASS=hassio
mosquitto_sub -h $MQTT_BROKER -u $MQTT_USER -P $MQTT_PASS -F "[\e[36m@x \e[96m@X.@N\e[39m] [\e[94mr%r:l%l\e[39m]\t\e[0m%t\t\e[1m%p\e[0m" -t "#"
```

proasnet, 2021-04-08:
Hello @aderusha . Now I am trying install mqtt command line clients, but in a terminal I get this message:
bash: sudo: command not found

I have Home Assistant as Hassio, standard instalation, not in a raspbian etc..
Please note that I am a beginner with HA, I am a programmer for industrial automation and I am an electronic engineer

proasnet, 2021-04-15:
I am sorry for open issue. Issue is still realy, next page on all HASPs is displayed after about 5 sec after pressing next button, but now I have not a free time for testing, I am working on long project in my work. For now, I am closing this, and I will open it later.
Thank you very much.

closed 2021-04-15 by proasnet