HASwitchPlate/HASPone #31: Find way, how change background color for each button on page6

proasnet, 2021-03-09:
On page 6 I have a call of color scenes of one RGBWW light. Now, I have here a texts with colors, but, is possible change background color of each button for easy recognize color preset?
For example:
White button without text for call white scene
Red button without text for call red scene
Blue button...etc...

I Find only this blueprint:
Display Toggle
But I'm not sure if this is the right choice

Thank you

aderusha, 2021-03-11:
Currently, the display text blueprint does not offer options for setting colors.  I'm setting this as a feature request, seems like something we could add to the blueprint!

aderusha, 2021-07-06:
Finally updated the [Display Text blueprint](https://github.com/HASwitchPlate/HASPone/blob/main/Home_Assistant/blueprints/hasp_Display_Text.yaml) with the ability to define the 4 color values manually or to use the current theme settings.

[image]

aderusha, 2021-07-06:
I've additionally published the [Apply Theme blueprint](https://github.com/HASwitchPlate/HASPone/blob/main/Home_Assistant/blueprints/hasp_Apply_Theme.yaml) which will allow setting theme colors to any buttons which don't have a blueprint handling that already.

[image]

closed 2021-07-06 by aderusha