HASwitchPlate/HASPone #32: Good tip for quick light

proasnet, 2021-03-09:
I tested idea for quick light in room. Instead of a motion sensor, I connected a capacitive sensor TTP223 to the D1, located under the HASP cover next to the display. My wife really appreciated that when she enters the room just for a little thing, she doesn't have to look for the command to light up on the display.
Very nice and simple. Good helper on white strong light !!!

aderusha, 2021-03-09:
One note - this should probably be over in [Discussions](https://github.com/HASwitchPlate/HASPone/discussions).  But I still want to see pictures, this sounds perfect :D

proasnet, 2021-03-09:
Thank you. Is very easy. I am sory for prototype looking, as you see, I have an enhanced version of Nextion and next him is touch sensor with TTP223

[image]

aderusha, 2021-03-09:
I'm going to close this as it's not really an issue w/ the project (but still a great idea!). Thanks for the tip :D

closed 2021-03-09 by aderusha