HASwitchPlate/HASPone #38: Custom Buttons Not Reloading After Reconnection to WiFi

IIAIronWolf, 2021-03-12:
I have an issue with my ISP currently that causes my WiFi to go out every now and then. To fix the issue, I have to restart my modem and router. 
I noticed that while the router was restarting, the HASP went back to AP mode, showing the QR code. When the WiFi reconnected, the custom buttons I had made for b(4) and b(6) of the alarm page had gone back to the Core Functionality defaults.
I had to manually run the automations to reset the buttons; however, I also noticed later that day that the buttons had yet again gone back to the core functionality defaults.

aderusha, 2021-03-15:
Can you explain a little bit about what you mean by custom buttons for b[4] and b[6]?  The core automation should only be touching buttons 1-3 on each page.  Do you have the alarm blueprint deployed?

IIAIronWolf, 2021-03-28:
I have deployed the alarm blueprint, but I don't like the new, non-colorized button for b(4) and b(6) that used to be in the old HASP demo. So I created my own custom buttons that change colors based on alarm state, like your old demo used to. However, when the HASP reconnects to wifi, those customization are lost.

aderusha, 2021-03-29:
Can you post the code you are using?  It sounds like you've deployed the Alarm blueprint, then also deployed some additional automations that interact with the same buttons.  You need to be pretty careful about that process for it to work, as you'll have two different sets of automations fighting for control over the same objects.  As a general rule, the last automation to fire "wins".

IIAIronWolf, 2021-04-13:
That being the case, it would seem the only way to get around this is to change the automation that your blueprint creates since there is currently no way to colorize the buttons?

aderusha, 2021-04-14:
Editing the blueprint locally is probably the best approach here.  It's why you don't see a lot of "all in one" blueprints published by me.  While several users have asked for this, I find having 12 buttons all defined in a single blueprint is bound to lead to issues like you've described.  You either take the blueprint as-is, or you're stuck changing the whole thing because you want to make a couple small changes.

It's the tradeoff that comes with a solution like this.

closed 2021-04-14 by IIAIronWolf