HASwitchPlate/HASPone #45: Feature: ability to set mqttDiscoveryTopic manually and override hardcoded value

scyto, 2021-03-15:
**Issue:**
Not everyone uses the default of homeassistant for the discovery topic (for example mine is hass-discovery).

**Ask**
Allow mqttDiscoveryTopic to be optionally set via the web interface (it is ok for it be autopopulated with homeassistant or whatever)

**Other solutions I have considered**
reconfiguring all services i have connected to MQTT to use homeassistant

aderusha, 2021-07-06:
Finally getting around to coding this request!  The HA discovery topic is now configurable in the web UI and changes are persisted to flash.  This feature will be in the upcoming [1.03 release](https://github.com/HASwitchPlate/HASPone/tree/1.03-HASPone-firmware).

If you want to test with the current dev release, run the web admin firmware update and paste in the following URL: `https://raw.githubusercontent.com/HASwitchPlate/HASPone/1.03-HASPone-firmware/Arduino_Sketch/HASwitchPlate.ino.d1_mini.bin`

aderusha, 2021-11-10:
[Fix released in HASPone v1.03](https://github.com/HASwitchPlate/HASPone/discussions/146)

closed 2021-11-10 by aderusha