HASwitchPlate/HASPone #48: Struggling to convert to NX4832T035_011R

OliPassey, 2021-03-18:
Hey,
I have two NX4832T035_011R, and one 2.4" all were running on the older version of HASP.
I've moved and got a new HA install on the go - tried to convert the vanilla nextion file into one that'll work with my NX4832T035_011R - but i get an error on model match.
I can't seem to select the NX4832T035_011R in nextion editor, only NX4832T035_011 shows up.

Do you know how i can amend this before compiling?

Looking forward to trying out the new version!

aderusha, 2021-03-29:
Hey sorry for the delayed response - did you ever find a resolution to this issue?

OliPassey, 2021-03-29:
No worries - no, sadly not yet.
A bit stumped on selecting the correct hardware - didn't have this issue last time round - wondering how i managed to flash them the first time round if the model numbers don't line up.

aderusha, 2021-03-29:
Yeah that has me a little stumped.  One suggestion I'll throw out - maybe try the [Unofficial Nextion Forums](https://unofficialnextion.com/).  We have a pretty diverse crowd there who might have a bit more experience with the specific model you're working with.  

The "official" support forums are a little... difficult to deal with, so I created this site for Nextion users to work together without needing to deal with the Nextion organization themselves.  If you've ever had the misfortune of doing so, you'll understand why this was (sadly) necessary :|

OliPassey, 2021-03-29:
Thanks, appreciate the steer.
I'll take another look at this asap and let you know how i get on.

fvanroie, 2021-04-18:
>  I can't seem to select the NX4832T035_011R in nextion editor, only NX4832T035_011 shows up.

The R is the product code shouldn't make any difference, did you try to compile and upload from the `NX4832T035_011` instead?

OliPassey, 2021-04-20:
Managed to get things working on what i suspect is the original tft file i created for the previous iteration of HASP. 

I have 2x 3.5's & 1x 2.4 - all are now working, but i messed up the font values on my file when i first created (kind of how i know the screens are still on the old firmware) i'm going to dig out the original HMI file, fix the font issues and re-compile.

Once i've finished i'll post the file here in case it's useful for anyone else.

@fvanroie - yes, thats what i compiled, and when uploading the screen showed a mismatch on just the _11R - i'll take screenshots if i can recreate it.

Either way, i have touchscreen controls back in my life - thanks for an awesome new version, loving the blueprint import function!

OliPassey, 2021-04-20:
Not sure if this was my mistake or the software being odd.
I've re-selected the device, recompiled and this time it's accepted the file.

I can't work out what was wrong as selected device didn't change anywhere visibly in the GUI.

Anyway, files for anyone who may want it / find it useful.
It's not perfect but it'll save someone the time of resizing everything.
You'll also want to amend the message on p0 :)

closed 2021-04-22 by OliPassey