HASwitchPlate/HASPone #51: Multi Gang Installation

omayhemo, 2021-03-20:
Unfortunately the included custom printed enclosure doesn't allow for mounting with-in a 2 or more gang.  The instructions also lack any detail about wiring.  Is there no ground, just neutral and line?

aderusha, 2021-03-20:
There are [several multi-gang models available](https://github.com/HASwitchPlate/HASPone/tree/main/3D_Printable_Models), and if you don’t find a variation that fits your needs, let me know and I can probably whip something up.

You are correct on the lack of ground, there are no exposed conductive surfaces so a separate ground is not needed.

omayhemo, 2021-03-20:
Thanks Allen,

I don't have a 3d printer.  Do you sell a 3 gang plate, with two paddles on one side, your HASPone on the other?

aderusha, 2021-03-21:
I do! For $15 I will print and ship a custom 3 gang plate to your specification.  Send $15 to paypal@svideo.org along with the specific layout you are looking for.  If you want HASP on the left, with a Decora ("paddle") in the middle and also on the right, send me something like "HASP | Decora | Decora" so I know exactly what you're after.

omayhemo, 2021-03-21:
Thanks Allen, Sent with note, but just in case
HASP | Decora | Decora

closed 2021-03-21 by omayhemo