HASwitchPlate/HASPone #64: landscape layout for Nextion Display 7-10 inch and more the same page in HASP

bluetech1-api, 2021-04-01:
I have a Nextion 7 inch and it look so not really well with portrait layout, so can I have fix some code to make it work as a landscape layout? I intend to buy a bigger one for nextion 10 inch, but if it can not work with landscape layout so it look so good. BTW, I should I do if I want more pages in HASP Page, example, I want 03 page 9 and 03 page 7 for all suitable my devices, is it possible?
Thanks for your help.

aderusha, 2021-04-01:
Both of the above are possible, but you're going to need to do a bit of work in the Nextion editor.  Also, the 10 inch model pulls more current than the power supply on the PCB can handle, so you'll either need external power or you'll need to develop your own PCB with footprint for a much larger display.

So, possible on both accounts, but not straightforward.  You'll have some work in front of you to make it happen.

closed 2021-04-01 by aderusha

bluetech1-api, 2021-04-02:
thanks aderusha, beside the power supply and a bit of work in the Nextion editor to resize the layout, so how about esp sketch and Home_Assistant/blueprints? where is I can begin exactly?

aderusha, 2021-04-02:
The ESP side should work fine, and the blueprints should work with the exception of the 4 "with icon" blueprints, which would need to be modified to place the overlay icon in the right spot.  That's mostly a cosmetic issue, and others have done similar work with other sizes of Nextion device.