HASwitchPlate/HASPone #66: Wifi Connected! screen keeps showing intermittently

omayhemo, 2021-04-02:
I've installed and configured the device.  Loaded the core blueprint and automation as well as configured a couple button automations.

Frequently the WiFi Connected! screen flashes up shows the SSID, IP Address and MQTT Connection IP, then within a second returns to the button screen.

aderusha, 2021-04-02:
What version of HASPone are you running?  If 1.00, can you try upgrading to 1.01 and see if the issue persists?

omayhemo, 2021-04-02:
1.01 resolved the periodic display of that WiFi view, however I can no longer connect to it via browser and IP.

omayhemo, 2021-04-02:
Spoke too soon, it's doing it again.  I'm guessing it's loosing wifi periodically and this screen displays when it connects?

aderusha, 2021-04-02:
That's correct, for some reason you're dropping off of WiFi.  Getting a debug log remotely via telnet can be tricky in these situations for obvious reasons. Somewhere near the bottom of the admin web page you should see a value for Signal Strength.  What is that reading right now?

Also, what kind of wifi APs are you running?  Are they able to tell you anything interesting about the event?

omayhemo, 2021-04-02:
Not great. -81db 
I know this room has weak wifi. So this much be the problem.

closed 2021-04-02 by omayhemo