HASwitchPlate/HASPone #72: jinja2.utils.Namespace object' has no attribute 'source'

damanti-me, 2021-04-10:
First of all thank you for your great work!
I just started with the project and received the following error in HA logs and noticed, that the blueprints didnt run through (the latest release of blueprints, noticed because i also didnt receive the YAML message after the described successfull runthrough of the automation assuming i turned on the option of course)
` Logger: `
`homeassistant.components.automation.hasp_p_x_b_y_displays_a_clock_with_a_clock_icon`
`Source: components/automation/__init__.py:452`
`Integration: Automatisierung (documentation, issues)`
`First occurred: xxxx`
`Last logged: xxxx`
`Error rendering variables: UndefinedError: 'jinja2.utils.Namespace object' has no attribute 'source'`
facts: mqtt working, touchscreen useable, no compilation error
A fried of mine replicated my setup with the privious HomeAssistant Version. Here its working. Seems to be a jinja issue with the new version. But im not sure.  Still some parts of the functionality is working (give the pages new names e.g.) with a hard reset on ESP side.
Thank you all for your time and patience

aderusha, 2021-04-10:
This turns out to be the result of [a breaking change in 2021.04](https://github.com/home-assistant/core/pull/48140).  I'm still trying to figure out how to address this and have reached out for some help [here](https://community.home-assistant.io/t/2021-04-breaking-changes-and-catching-run-actions/297977).

damanti-me, 2021-04-10:
fixed with commit aderusha@2ee34cb5d30deb3a9e725ea561ad717dd2d264b2

closed 2021-04-10 by damanti-me