HASwitchPlate/HASPone #79: Plate Stuck on Setup Screen

npeskoe, 2021-04-14:
I have successfully paired my plate with HA and have downloaded the blue prints. I have tried to run automations to get the device fully set up and functioning, but it is stuck on the Wifi Connected/MQTT Connected Screen. I have tried rebooting the device and updating the firmware, but am not having any luck. Can someone please assist?

Thanks

damanti-me, 2021-04-15:
is there a mqtt error code on the display?
what does your broker shows in the log?

npeskoe, 2021-04-15:
No error on the display. It says wifi and mqtt are both connected.

This is the log when I reboot the device:

[08:32:53] WARNING: [WARN] Not found homeassistant on local database
1618489973: Socket error on client <unknown>, disconnecting.
1618490256: New connection from 172.30.32.1 on port 1883.
1618490256: Socket error on client <unknown>, disconnecting.
1618490258: New connection from 172.30.32.1 on port 1883.
1618490258: Socket error on client <unknown>, disconnecting.
1618490260: New connection from 172.30.32.1 on port 1883.
1618490260: Socket error on client <unknown>, disconnecting.
1618490264: New connection from 172.30.32.1 on port 1883.
1618490264: Socket error on client <unknown>, disconnecting.
1618490272: New connection from 172.30.32.1 on port 1883.
1618490272: Socket error on client <unknown>, disconnecting.
1618490288: New connection from 172.30.32.1 on port 1883.
[08:38:09] WARNING: [WARN] Not found homeassistant on local database
1618490289: Socket error on client <unknown>, disconnecting.
1618490321: New connection from 172.30.32.1 on port 1883.
1618490321: Socket error on client <unknown>, disconnecting.
1618490385: New connection from 172.30.32.1 on port 1883.
1618490385: Socket error on client <unknown>, disconnecting.
1618490496: Client peskoeplate-e8db84c5e5db disconnected.
1618490498: New connection from 192.168.1.233 on port 1883.
[08:41:39] INFO: [INFO] found npeskoe on Home Assistant
1618490499: New client connected from 192.168.1.233 as peskoeplate-e8db84c5e5db (p2, c1, k30, u'npeskoe').
1618490505: New connection from 172.30.32.1 on port 1883.
1618490505: Socket error on client <unknown>, disconnecting.
1618490625: New connection from 172.30.32.1 on port 1883.
[08:43:46] WARNING: [WARN] Not found homeassistant on local database
1618490626: Socket error on client <unknown>, disconnecting.
1618490746: New connection from 172.30.32.1 on port 1883.
1618490746: Socket error on client <unknown>, disconnecting.
1618490866: New connection from 172.30.32.1 on port 1883.
1618490866: Socket error on client <unknown>, disconnecting.

damanti-me, 2021-04-15:
Try this link@https://community.home-assistant.io/t/there-is-no-homeassistant-local-user-in-the-database/204207

npeskoe, 2021-04-15:
Thank you very much. @damanti-me That fixed it.

closed 2021-05-16 by aderusha