HASwitchPlate/HASPone #82: Non Bug Topic: Set up GitHub Sponsors?

psarossy, 2021-04-17:
Allen, the work you are doing is awesome, and I see that you have a buymeacoffee.com setup, but you may want to consider setting up GitHub Sponsors for this repo, folks seem to appreciate the work going into the project and automated donations tend to work much better than one offs that one has to remember to do...

I for would would definitely chip in... (would just recommend to set up a few tiers, and not scare folks with a too high minium)

aderusha, 2021-04-17:
Sounds like a good idea!  Hardware development carries a cost, and every little bit helps me out :D

Github needs to approve my account and I'll update when that happens, thanks for the idea!

aderusha, 2021-04-17:
Allright, looks like we're live!

fvanroie, 2021-04-18:
You also need to create a [.github/FUNDING.yml](https://github.com/HASwitchPlate/openHASP/commit/964a0f40c1c3175221b238934977eb23c581f91f) before a sponsor button shows up on top of the repository page.
I added your GitHub Sponsor account and buymeacoffee link to openHASP as well.

IIAIronWolf, 2021-04-28:
Great idea!

closed 2021-05-16 by aderusha