HASwitchPlate/HASPone #90: Errors produced during setup. Non functioning panel.

Is0mniac, 2021-04-29:
Hello, I bought a new Haspone from you, and I am having multiple errors during setup. It connects to my mqtt server (run on external rasp pi, home assistant is running as the virtual env package for synology), and connects to my wifi. Deploying the core integration, however gives me nothing, the device doesn't reboot, stays on WiFi connected, and Mqtt Connected screen. It doesn't reboot as stated in the instructions. If I touch the panel it does take me to an updated version of the page views, and I am able to see the page buttons below, but nothing for buttons 4-7 on any of the pages. Deploying a blueprint for anything fails to run. I am getting multiple errors in my logs. The buttons all remain blank except the bottom navigation labels.

homeassistant.components.automation.new_automation
Source: components/automation/__init__.py:452
Integration: Automation (documentation, issues)
First occurred: 5:35:07 AM (86 occurrences)
Last logged: 5:51:07 AM

Error rendering variables: UndefinedError: None has no element 0

homeassistant.components.automation.hasp_core_functionality_3
Source: helpers/script.py:1324
Integration: Automation (documentation, issues)
First occurred: 5:37:31 AM (2 occurrences)
Last logged: 5:37:31 AM

HASP Core functionality: Choose at step 1: choice 1: Error executing script. Error for call_service at pos 10: Error rendering data template: UndefinedError: None has no element 0
HASP Core functionality: Error executing script. Error for choose at pos 1: Error rendering data template: UndefinedError: None has no element 0

homeassistant.components.automation.hasp_core_functionality_3
Source: components/automation/__init__.py:504
Integration: Automation (documentation, issues)
First occurred: 5:37:31 AM (1 occurrences)
Last logged: 5:37:31 AM

Error while executing automation automation.hasp_core_functionality_3: Error rendering data template: UndefinedError: None has no element 0

I am not sure if this is related to home assistant updates or some other issue. Can you please guide me on how to fix this?

Thank you so much!

raa1n, 2021-04-29:
I have the exact issue except that I built the HASP myself instead of buying one directly. I can see that it is connected to my wifi and MQTT connection is established.

When I run the core functionality blueprint, I get the same error logged above and the HASP display doesn't change either

Thanks in advance for the solution (hopefully..!!!)

aderusha, 2021-05-03:
@Is0mniac and @raa1n -[ check out this post](https://github.com/HASwitchPlate/HASPone/discussions/93#discussioncomment-686020) for what I think should be a solid solution.  If you update, let me know how it goes and if everything looks good I'll push this release to the main branch.

Is0mniac, 2021-05-03:
It works!! Thank you much.

aderusha, 2021-05-16:
@raa1n how are things looking for you buddy?  The test version I posted above is now release v1.02.  Is everything working on the new codebase?

aderusha, 2021-06-18:
Closing as OP has been resolved.

closed 2021-06-18 by aderusha