HASwitchPlate/HASPone #95: Blueprint HASP p[x].b[y] displays the current temperature with icon and colors fails with temperature sensor

rebel7580, 2021-05-03:
When creating an automation from this blueprint and selecting a temperature sensor instead of a weather provider, the automation fails. When saved, no "Run" is shown, and the automation does not show in the Automation list. The automations do show up in automations.yaml.

The homeassitant log shows the following ERRORs (tried twice with two different temperature sensors):

2021-05-03 16:19:26 ERROR (MainThread) [homeassistant.components.automation] Blueprint HASP p[x].b[y] displays the current temperature with icon and colors generated invalid automation with inputs OrderedDict([('haspdevice', 'cfc57889d19f2ce169bf924e4d916c67'), ('hasppage', 2), ('haspbutton', 5), ('temperature_sensor', 'sensor.outside_temp'), ('colortemp', True)]): Entity IDs can not be None for dictionary value @ data['entity_id']. Got None

2021-05-03 16:19:27 ERROR (MainThread) [homeassistant.components.automation] Blueprint HASP p[x].b[y] displays the current temperature with icon and colors generated invalid automation with inputs OrderedDict([('haspdevice', 'cfc57889d19f2ce169bf924e4d916c67'), ('hasppage', 2), ('haspbutton', 4), ('temperature_sensor', 'sensor.hotwater_am2301_temperature'), ('colortemp', True)]): Entity IDs can not be None for dictionary value @ data['entity_id']. Got None

When choosing a weather provider instead, the automation is successful and is shown inthe Automation list.

I know almost zero about blueprint/automation coding, but these lines in hasp_Display_Temperature_with_Icon_and_Colors.yaml look suspicious to me:

```
381: trigger:
382: - platform: state
383:   entity_id: !input 'weather_provider'
```

rebel7580, 2021-05-06:
I'm definately a noobie when it comes to HA and yaml, so take this FWIW:

There seems to be two issues here:
1) the trigger for weather_provider will fail if no weather provider is defined,
2) there is no trigger for the temperature sensor, if defined. 

I tried this
```
trigger_variables:
   ...
  weather_provider: !input 'weather_provider'
  temperature_sensor: !input 'temperature_sensor'
  entityid: '{%- if temperature_sensor|lower == "none" -%}  {{- weather_provider -}} {%- else -%}  {{- temperature_sensor -}} {% endif %}'
trigger:
- platform: state
  entity_id: '{{ entityid }}'
```
but this causes an error also, probably based on mal-formed yaml on my part. The idea was to set 'entity_id' to either the weather entity or the temperature entity.

Of course, I may completely misunderstand this code!

rebel7580, 2021-05-19:
Since those new trigger variables didn't work, I reverted back and then as an experiment, simply changed the trigger entity like this:
```
381: trigger:
382: - platform: state
383:   entity_id: !input 'temperature_sensor'
```
This allowed temperature sensor entities to work assuming one was entered in the blueprint, but of course weather ones now don't.

aderusha, 2021-07-08:
I've done some digging into this and based on conversations with some of the Nabu Casa folks.. what I'm trying to do there cannot be done.  Triggering on entity state change with a blueprint must be in the format of `entity: !input <some_input_entity>`.  That means you can't make decisions before the trigger condition as to which entity should be used, and you can't trigger on both unless you force the user to assign both.

I'm not quite sure how to proceed on this one.  One solution might be to revert this blueprint back to a weather-only solution, then to create a second blueprint which allows a sensor instead of a weather entity.  And if I do that... maybe it could be more general, such that one can define 5 different icons and ranges which default to the temperature icons (but allow the user to substitute in any other icons).  That way, one could use it with any sensor to display the current value and have colors+icons along for the ride.

rebel7580, 2021-07-09:
> I've done some digging into this and based on conversations with some of the Nabu Casa folks.. what I'm trying to do there cannot be done. Triggering on entity state change with a blueprint must be [...]

Well, at least I don't feel too dumb not being able to figure out how to do it! :)

> I'm not quite sure how to proceed on this one. One solution might be to revert this blueprint back to a weather-only solution, then to create a second blueprint which allows a sensor instead of a [...]

That's s good idea, one for weather, and one similar for general sensor.

In my case, since I use HA only as an "add-on" to do a few things that are hard for my own home automation system to do, I've dropped using the Hasp blueprints and implemented similar functionallity in my system (and modeled some of the logic from the blueprints). That way, I have fewer points of possible failure for critical things.  

I have several inside and outside temp/hum sensors that display and look a lot like yours. For the outside ones, I've also include a "feels like" temp display based on temp and humidity and wind speed. I also color that one with a range of blue-green-yellow-orange-red depending on value.  It's been mostly red these last few days! I plan to add the 5-range temp icons soon (right now it's just showing the max temp icon).

Thanks for this great project!

rebel7580, 2021-09-07:
Should this be closed via commit: https://github.com/HASwitchPlate/HASPone/commit/8194fc273bfdf46425f21e6e48fc103503140e9c
which removed temp sensor from selection? Or are you waiting until/if you do a temp sensor version before closing this one?

aderusha, 2021-09-08:
> are you waiting until/if you do a temp sensor version before closing this one?

Yup, just need to get off my duff and finish up the temp sensor version, and I'm leaving this open to annoy me until I do that :D

aderusha, 2021-11-18:
OK I think I have this sorted: https://github.com/HASwitchPlate/HASPone/discussions/152

closed 2021-11-18 by aderusha