HASwitchPlate/HASPone #98: Arduino Sketch Build Issues

blaxer, 2021-05-07:
I am struggling to get the Arduino sketch to build properly, the link to the documentation for the ardunio sketch on this page: 

https://github.com/HASwitchPlate/HASPone/tree/main/Arduino_Sketch 

is broken and the documentation here: 

https://github.com/aderusha/HASwitchPlate/blob/master/Documentation/01_Arduino_Sketch.md 

appears to be out of date. 

I suspect that ESP8266 version 2.5.0 may no longer be the correct board version? In addition, ESP8266Ping does not appear to be avilable in the library manager but this might be the library? 

https://github.com/dancol90/ESP8266Ping

Any information around how to succesfully build the sketch would be very much appreciated! TIA!

aderusha, 2021-05-15:
If you use platformio, here's what I have in my platformio.ini:

```ini
[env:d1_mini]
platform = espressif8266
board = d1_mini
framework = arduino
board_build.f_cpu = 160000000L
board_build.ldscript = eagle.flash.4m1m.ld
build_flags = 
  -D PIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_191122 
  -D PIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH
monitor_filters = esp8266_exception_decoder
lib_deps = 
  ArduinoJson
  MQTT
  dancol90/ESP8266Ping@^1.0
  krzychb/EspSaveCrash@^1.2.0
```

aderusha, 2021-06-18:
@blaxer are you all set?

shafr, 2021-06-23:
Also libs I used was:
```
lib_deps =
      krzychb/EspSaveCrash @ ^1.2.0
      tzapu/WiFiManager @ ^0.16.0
      bblanchon/ArduinoJson         @ ^6.18.0     
      256dpi/MQTT @ ^2.5.0
      dancol90/ESP8266Ping @ ^1.0
```

saikek, 2021-07-01:
> Also libs I used were:
> 
> ```
> lib_deps =
>       krzychb/EspSaveCrash @ ^1.2.0
>       tzapu/WiFiManager @ ^0.16.0
>       bblanchon/ArduinoJson         @ ^6.18.0     
>       256dpi/MQTT @ ^2.5.0
>       dancol90/ESP8266Ping @ ^1.0
> ```

aderusha, 2021-07-06:
I'm now going to start including a working `platformio.ini` in future firmware build updates to allow folks to pull the correct versions of any included libraries to make building the project a bit easier.  Currently, that file looks like this on the 1.03 dev branch:

```ini
[env:d1_mini]
platform = espressif8266
board = d1_mini
framework = arduino
board_build.f_cpu = 160000000L
board_build.ldscript = eagle.flash.4m1m.ld
build_flags = 
    -D PIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_191122 
    -D PIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH
lib_deps = 
    bblanchon/ArduinoJson @ ^6.18.1
    256dpi/MQTT @ ^2.5.0
    dancol90/ESP8266Ping @ ^1.0
    krzychb/EspSaveCrash @ ^1.2.0
    https://github.com/tzapu/WiFiManager.git#8984863fe6a8106bf0ead7e6ac013c745a4d1f93
```

closed 2021-07-06 by aderusha

fvanroie, 2021-07-06:
It also allows you to set up a CI/CD pipeline where Github Actions will automatically build the firmware from source on every commit, etc...