HASwitchPlate/HASPone #99: Dropped config on restart

OliPassey, 2021-05-07:
I've got a few HASPone units going, but they seem to drop their config on a reboot of HA or the device itself.
I have to re-run automations to get things looking right again, this used to be automatic - have i missed a step?

Any advice you can offer would be greatly appreciated.

aderusha, 2021-05-15:
What configuration is being dropped?  The configuration of the device itself (the items seen on the web admin page), or blueprint config?  Something else?

OliPassey, 2021-05-15:
Sorry, should have been more clear.
All the text labels and button presses I've added are not sent to the device and I have to manually rerun my automation to populate the device again. Otherwise it connects and just has blank buttons on all pages.

aderusha, 2021-05-15:
Can you take a look at [this procedure](https://github.com/HASwitchPlate/HASPone/discussions/102#discussioncomment-741804) and see if that resolves anything?

OliPassey, 2021-05-16:
No duplicate devices have been created, but in troubleshooting today i saw v1.0.2 was available.
Upgrading has seemed to solve it, i can now reboot the device and they'll automatically populate as before.

Not sure what the exact problem ended up being - but hey - working now.

Thanks for the advice 👍

closed 2021-05-16 by OliPassey