HASwitchPlate/HASPone #105: Odd json state return when querying text attribute

rebel7580, 2021-06-14:
I'm using the hasp with my own MQTT based home automation system.  I do use the default display sw. 
HASP Version: 1.02
LCD Version: 3

Right now, I just have page 1 defined with actual button values. If I query the txt attribute, I not only get a state button response with the correct text, but also get a state json response from p[7].b[15] with the same text. This is the log from my system:
```
12:06:35.991 mqttx: Published hasp/hasp01/command/p[1].b[7].txt, !
12:06:36.001 mqttx: Received PUBLISH (d0, q0, r0, 'hasp/hasp01/state/p[1].b[7].txt', ... (9 bytes))
12:06:36.002 hasp: hasp: topic:hasp/hasp01/state/p[1].b[7].txt,msg:Overcast ;
12:06:36.008 mqttx: Received PUBLISH (d0, q0, r0, 'hasp/hasp01/state/json', ... (46 bytes))
12:06:36.009 hasp: hasp: topic:hasp/hasp01/state/json,msg:{"event":"p[7].b[15].val","value":"Overcast "};
```
You can see the correct "Overcast " response for p[1].b[7], but the json is reported mistakenly from p[7].b[15]!

Here is the response if I query p[7].b[15] directly:

```
12:06:42.265 mqttx: Published hasp/hasp01/command/p[7].b[15].txt, !
12:06:42.273 mqttx: Received PUBLISH (d0, q0, r0, 'hasp/hasp01/state/p[7].b[15].txt', ... (0 bytes))
12:06:42.274 hasp: hasp: topic:hasp/hasp01/state/p[7].b[15].txt,msg:;
12:06:42.278 mqttx: Received PUBLISH (d0, q0, r0, 'hasp/hasp01/state/json', ... (37 bytes))
12:06:42.280 hasp: hasp: topic:hasp/hasp01/state/json,msg:{"event":"p[7].b[15].val","value":""};
```
You can see that both the state button and state json responses are the correct empty value.

Sending queries to page 1 buttons 4-7 yield the same response from page 7 button 15.

But, further testing shows that querying button 4 on any page (or any other valid button for that page)  returns the incorrect page/button for the json return (but the correct attribute value).  So while the second log above seems to show a correct json response for page 7 button 15, that's just a coincidence since queries to ALL page have in the response page 7 button 15 as the button address.  And it doen't always return p[7].b[15] in the json. the actual page/button seems to change based on query history and page selections. At one point during my testing of this issue, the json return changed to p[11].b[5].

Possibly, there is some incorrect "memory" somewhere that is being used as the json page/button value in the return (although the actual text attribute value is returned properly).

aderusha, 2021-07-06:
Finally getting around to digging into this and I am unable to reproduce this behavior using the steps provided and I'm hoping you can help me out.

## MQTT log

```
1. hasp/plate01/command/page       8
2. hasp/plate01/state/json {"event":"page","value":8}
3. hasp/plate01/state/page 8
4. hasp/plate01/state/p[8].b[4]    ON
5. hasp/plate01/state/json {"event_type":"button_short_press","event":"p[8].b[4]","value":"ON"}
6. hasp/plate01/state/p[8].b[4]    OFF
7. hasp/plate01/state/json {"event_type":"button_short_release","event":"p[8].b[4]","value":"OFF"}
8. hasp/plate01/state/p[8].b[4].val        0
9. hasp/plate01/state/json {"event":"p[8].b[4].val","value":0}
10. hasp/plate01/command/p[1].b[4].txt
11. hasp/plate01/state/p[1].b[4].txt        Text to display test 1 2 3
12. hasp/plate01/state/json {"event":"p[8].b[4].val","value":"Text to display test 1 2 3"}
```

## Explanation:

**1:** I send MQTT command to switch to page 8
**2-3:** HASPone replies with current page on JSON and state topics
**4-7:** At the device I press and then release `p[8].b[4]` and the HASPone send the `ON` and `OFF` messages on JSON and state topics
**8-9:** HASPone sends current value of `p[8].b[4]`
**10:** I send MQTT message querying the value of `p[1].b[4].txt`
**11-12:** HASPone replies with current value of `p[1].b[4].txt` on JSON and state topics

rebel7580, 2021-07-06:
Actually, your test shows the problem!  Note step 12 json event reports using p[8].b[4].val, not p[1].b[4].val like step 11, although the actual value IS correct for p[1].b[4].

Tried this again with same results as before.  
Note:  "mqttx" is my mqtt client log, "hasp" is my hasp app log.
```
SEND PAGE 3 COMMAND:
06/11:24:06.866 mqttx: Published hasp/hasp01/command/page, 3!

RECEIVE RESPONSE:
06/11:24:06.920 mqttx: Received PUBLISH (d0, q0, r0, 'hasp/hasp01/state/json', ... (26 bytes))
06/11:24:06.921 hasp: hasp: topic:hasp/hasp01/state/json,msg:{"event":"page","value":3};
06/11:24:06.922 mqttx: Received PUBLISH (d0, q0, r0, 'hasp/hasp01/state/page', ... (1 bytes))
06/11:24:06.923 hasp: hasp: topic:hasp/hasp01/state/page,msg:3;

PRESS BUTTON 3,4 (no content! i.e., not programmed with a text value.) 
06/11:24:13.679 mqttx: Received PUBLISH (d0, q0, r0, 'hasp/hasp01/state/p[3].b[4]', ... (2 bytes))
06/11:24:13.680 hasp: hasp: topic:hasp/hasp01/state/p[3].b[4],msg:ON;
06/11:24:13.686 mqttx: Received PUBLISH (d0, q0, r0, 'hasp/hasp01/state/json', ... (68 bytes))
06/11:24:13.687 hasp: hasp: topic:hasp/hasp01/state/json,msg:{"event_type":"button_short_press","event":"p[3].b[4]","value":"ON"};

MY APP RESPONDS BY LIGHTING THE DISPLAY:
06/11:24:13.689 mqttx: Published hasp/hasp01/brightness/set, 100!
06/11:24:13.690 mqttx: Published hasp/hasp01/light/switch, ON!

HERE'S THE OFF RESPONSE:
06/11:24:13.995 mqttx: Received PUBLISH (d0, q0, r0, 'hasp/hasp01/state/p[3].b[4]', ... (3 bytes))
06/11:24:13.996 hasp: hasp: topic:hasp/hasp01/state/p[3].b[4],msg:OFF;
06/11:24:14.000 mqttx: Received PUBLISH (d0, q0, r0, 'hasp/hasp01/state/json', ... (71 bytes))
06/11:24:14.001 hasp: hasp: topic:hasp/hasp01/state/json,msg:{"event_type":"button_short_release","event":"p[3].b[4]","value":"OFF"};

06/11:24:14.003 mqttx: Received PUBLISH (d0, q0, r0, 'hasp/hasp01/state/p[3].b[4].val', ... (1 bytes))
06/11:24:14.004 hasp: hasp: topic:hasp/hasp01/state/p[3].b[4].val,msg:0;
06/11:24:14.007 mqttx: Received PUBLISH (d0, q0, r0, 'hasp/hasp01/state/json', ... (35 bytes))
06/11:24:14.008 hasp: hasp: topic:hasp/hasp01/state/json,msg:{"event":"p[3].b[4].val","value":0};

REQUEST TXT VALUE FOR 1,7 ("Clear"):
06/11:24:21.063 mqttx: Published hasp/hasp01/command/p[1].b[7].txt, !
06/11:24:21.074 mqttx: Received PUBLISH (d0, q0, r0, 'hasp/hasp01/state/p[1].b[7].txt', ... (6 bytes))
06/11:24:21.075 hasp: hasp: topic:hasp/hasp01/state/p[1].b[7].txt,msg:Clear ;
06/11:24:21.076 mqttx: Received PUBLISH (d0, q0, r0, 'hasp/hasp01/state/json', ... (42 bytes))
06/11:24:21.077 hasp: hasp: topic:hasp/hasp01/state/json,msg:{"event":"p[3].b[4].val","value":"Clear "};
```
Note Event report has wrong button designation - 3,4 instead of 1,7 - but reports 1,7 value correctly.

Tried with a button that had been previously set to a txt value, and same thing happens.
The event report always uses the button address of the last button pressed, not the one aksed for.

aderusha, 2021-07-06:
Oh man I can't believe I missed that, thanks for the catch!  Got it, now I have the test case so I suspect I need to fix it :D

Thanks @rebel7580 , ball is squarely in my court.

aderusha, 2021-07-06:
Now that I know what the heck I'm looking at (thanks to the gentle guidance from @rebel7580) I think I've resolved the issue.  Dumb error in the firmware, and I believe I've identified and fixed the issue in the [1.03 branch](https://github.com/HASwitchPlate/HASPone/tree/1.03-HASPone-firmware).  Testing now returns the correct object ID both in the standard namespace and in the JSON object returned.

If you want to test with the current dev release, run the web admin firmware update and paste in the following URL: `https://raw.githubusercontent.com/HASwitchPlate/HASPone/1.03-HASPone-firmware/Arduino_Sketch/HASwitchPlate.ino.d1_mini.bin`

rebel7580, 2021-07-06:
Tested your dev build. Looks fixed! Thanks!

aderusha, 2021-11-10:
[Fix released in HASPone v1.03](https://github.com/HASwitchPlate/HASPone/discussions/146)

closed 2021-11-10 by aderusha

rebel7580, 2021-11-10:
Hi Allen -
Is the official 1.03 the same as the 1.03 branch mentioned on the 6 Jul post? Wondering if I need to upgrade if I already have the Jul branch version.

aderusha, 2021-11-10:
It has been updated since then, so you'll want to apply the release version.  If you go to the firmware update page and just let it use the default URL, you'll be applying the 1.03 release version.

rebel7580, 2021-11-10:
Updated my two hasps. Thanks!

aderusha, 2021-11-10:
Thank you for all the helpful feedback and bug hunting!