HASwitchPlate/HASPone #106: Very concerned about the many warninings in log wile useing Heatpump blueprints.

ssirkakriss, 2021-06-17:
I've been using HASwitchPlate/HASPone  for a bout 2 months. It works great for me, except for one blueprint. The heatpump blueprints. The blueprint seems to work fine, except it produces a lot of warnings in the log. I'm concerned after an upcoming update to HA in Oct. well turn the warnings in to errors. Link to update info: https://www.home-assistant.io/blog/2021/04/07/release-20214/#warnings-for-undefined-variables-in-templates.

The warnings I see in HA log.:
~~~
Logger: homeassistant.helpers.template
Source: helpers/template.py:1366
First occurred: June 13, 2021, 8:53:05 AM (2190 occurrences)
Last logged: 7:34:33 AM

Template variable warning: 'hp_is_off' is undefined when rendering '{%- if hp_is_off -%}heat_cool{%- else -%}off{%- endif -%}'
~~~

Is this something to worry about? 
Can the warnings be remedied?
I not a good coder but I get the gist of what the blog post is telling me. Sort of.

aderusha, 2021-06-22:
You are spot-on with your assessment of the situation.  This is OK for now, but won't be in October.  I don't have a heat pump here to test on, but am reaching out to the original developer of this blueprint for assistance in addressing the undeclared variable issues.

If anyone else has a heatpump setup, feel free to jump in here :D

aderusha, 2021-06-23:
@ssirkakriss - @SittingDuc has issued a PR for the page 6 heatpump blueprint.  Are you using page 6 or page 9?  If page 6, can you test to see if this resolves the issue?

SittingDuc, 2021-06-23:
Hello, my pr actually touched both page6 and page9, so either and both
should be happier/warning-free now. Would love to hear if it fixes the
issue for you

ssirkakriss, 2021-06-23:
I've removed the old blueprint and associated automation from my setup(s) and applied SittingDuc's PR (/SittingDuc/HASPone/blob/f092966202d452dcd508a5390fb3126b9c6f3c71/Home_Assistant/blueprints/hasp_Display_Heatpump_Control_page6.yaml. )
Note: Ive not tried Page 9 BP. As My setup(s) use page 6 BP.

I'm happy to report issue appears resolved. 
Many Thanks to all.

closed 2021-06-23 by ssirkakriss