HASwitchPlate/HASPone #108: Page scrolling mode.  Page select button 2 still active

sco0ts, 2021-06-18:
When core function is set to  "Page scrolling mode".  The middle button which was previously "Page select button 2 page" still goes to the page selected in the config when the middle button is pressed pressed.  Not sure if this is intentional

aderusha, 2021-06-18:
Yessir, that is intended functionality.  Gives you a quick-select page that you can go to with a single button press.  Didn't really have any other use for that in scroll mode so seemed like a reasonable thing to do.  I'm open to suggestions if you have a better idea!

sco0ts, 2021-06-18:
I think the logic is great I just wanted to make sure it was intentional.  A user may want to disable the middle button press but I dont think its necessary to add .

closed 2021-06-18 by aderusha