HASwitchPlate/HASPone #109: hasp_Dim_Screen_on_Idle.yaml - anyway to turn off screen?

sco0ts, 2021-06-21:
Thank you for producing hasp_Dim_Screen_on_Idle.yaml.  I was curious if there was a way to have the screen go to 0 (or ideally off) on idle?

aderusha, 2021-06-22:
There wasn't, but now that you said that, I think it's a good idea (thanks!) and I've [updated the blueprint](https://github.com/HASwitchPlate/HASPone/blob/main/Home_Assistant/blueprints/hasp_Dim_Screen_on_Idle.yaml) so that you can now enter `0` as a brightness value for the dimming, which will turn off the backlight entirely. 

Great idea and thanks for the issue!

aderusha, 2021-06-22:
Bedroom installations are the reason I wrote the blueprint in the first place :D  Glad this seems to fit your use case!

ThisIsTheOnlyUsernameAvailable, 2021-07-15:
> This is wonderful. I am going to use this for a neat use case (IMO). I have bedside HASPone's that during the day will dim to 20 when idle and go up to 60 when pressed. At night, when all the lights are off, locks are locked, and security system is active screen will dim to 0 when idle and up to 40 when in use. This has very very high wife approval. Thank you for making us both very happy!

Very cool.  I've love to do something similar.  How do you set the variable dimming?

sco0ts, 2021-07-15:
There a probably many ways to do this and most are probably better than what I came up with.  I used one template and four automations.  The template creates a binary sensor that if on indicates we are ready for 'night mode.' Then two automations that derived from the Dim screen on idle blue print.  One for the day state one for the night.  Then two more automation triggered by the state of the template, that execute the day or the night automation mentioned above.

I've attached the two as pasting to this message board seems to mess up the formatting
[Automations.txt](https://github.com/HASwitchPlate/HASPone/files/6823752/Automations.txt)
[Template.txt](https://github.com/HASwitchPlate/HASPone/files/6823755/Template.txt)