HASwitchPlate/HASPone #134: Error: Template warning: 'round'

xHirscHx, 2021-10-09:
Im getting this.

Logger: homeassistant.helpers.template
Source: helpers/template.py:1210
First occurred: 5:14:30 PM (3 occurrences)
Last logged: 5:14:30 PM

Template warning: 'round' got invalid input 'None' when rendering template '{%- if roundtemp == true -%} {{- state_attr(weather_provider, "temperature") | round -}} {%- else -%} {{- state_attr(weather_provider, "temperature") -}} {%- endif -%}' but no default was specified. Currently 'round' will return 'None', however this template will fail to render in Home Assistant core 2021.12

aderusha, 2021-11-10:
Thanks for the head's up @xHirscHx !  I've pushed updates to all blueprints which should address this warning in the log.  [Here's some info on that](https://github.com/HASwitchPlate/HASPone/discussions/143).

If you don't update, the device should still function but will continue to log errors.  The updated blueprints should avoid those log messages.

aderusha, 2021-11-18:
Closing issue after releasing fixes to impacted blueprints

closed 2021-11-18 by aderusha