HASwitchPlate/HASPone #141: HASPOne page 7 (Alarm)

ekrotki, 2021-11-01:
Hello, I have found that automation created with blueprint for page 7 is able to arm / disarm the alarm without entering the code. More precisely saying - last entered code is used. For instance, when I do arm the alarm using correct code, it is enough to hit "disarm" to get the alarm disarmed. Obviously entering wrong code before hitting "disarm" does not do anything.
Same is for "Arming" - if the alarm was disarmed using good code, it is enough to hit "Arm" and the alarm is armed.
I feel that the code is cached / stored and not cleared after hitting the "operation button".
Can we get it cleared after hitting the action button?
Another idea - can we have several codes for few people at home? Say - per user code?
Please don't shoot if I missed something :-)

aderusha, 2021-11-02:
Well that shouldn't be happening :(

If you [take a look at the blueprint code](https://github.com/HASwitchPlate/HASPone/blob/main/Home_Assistant/blueprints/hasp_Display_Alarm_Control_page7.yaml#L451-L458), what should be happening is that the current code entry should be submitted to home assistant and then immediately reset (all digits set to `-1`).

That leaves the question as to why that isn't happening for you.  Can you monitor the state of each of the `input_number.` entities while you enter the code and then select arm/disarm?  It'd be interesting to know if/how those are cycling when you perform an arm or disarm at the panel.

ekrotki, 2021-11-02:
Yes, you are right, unbelievable, but it happens. Regarding monitoring the numbers, well, what I can do is looking at "Overview" page and the numbers "alarmcode 1-6", they are filled in, shifting with each key-press, then after I hit "Arm", after some time the alarm panel shows being armed, numbers remain uncleared. For your information, I am using "manually" created alarm panel because I don't have real unit integrated, just wanted playing for the future application. This could be the factor if I did something wrong defining the "alarm panel".
I am also trying to get "kids protection" for wall plates, sometimes I do have guests with kids pushing every button in the reach of their hands... Something like "lock/unlock" the wall-plate using PIN.

ekrotki, 2021-11-07:
> Well that shouldn't be happening :(
>
> If you [take a look at the blueprint [...]

I have found that setting the number to "-1" is not allowed - tried with following call using Developer tools and failed.
> service: number.set_value
> target:
>   entity_id: number.plate01_alarmcode_1
> data:
>   value: '-1'

Trying positive value it goes successfully. Now the question is how to reset the values, or how to change the validation allowing -1 at least (if not all negative values).

aderusha, 2021-11-07:
@ekrotki Have you upgraded your install to 2021.11 by chance?

ekrotki, 2021-11-07:
@aderusha Yes, I am on 2021.11

I see the lines 321,322 now allowing the values -1 up to 9 solving the problem, Code numbers are reset to -1 correctly.
Now I started thinking about binding the reset operation with state change and not just b15 button press, IMHO it would create nice feature allowing quickly revert the operation. Codw would be cleared after the delay needed for alarm to change the state.

Do you see any chance to support multiple codes? In real burglary unitwe do have at home, each user has own code granting different permissions there. Definitely it is beyond the issue (solved now) but it might be welcomed by others.
thank you for your good work !!

closed 2021-11-07 by ekrotki

aderusha, 2021-11-08:
Great news, glad we got it working!  

Almost everything about how the alarm system operates is dictated by the alarm device.  HASPone (with this blueprint) is simply submitting codes when entered, it's up to the alarm panel to decide what to do with them.  So, HASPone for sure supports multiple codes, which is because it doesn't actually deal with any codes at all - it's simply accumulating whatever you enter, then sending the last *n* digits to your alarm device when you press arm/disarm.

ekrotki, 2021-11-08:
@aderusha Yes, I am glad too. With apologies of posting here again - I am not yet well acquainted where I shall do it :-)
My quick solution was to use "16" instead of -1, but this is also OK.

Yes, keeping HASPone clean from "semantic" is best way to go - it shall only be the "I/O interface"  or HMI. All else is the task of something listening to MQTT.
This is why I think about implementing "screen-lock" via switching "alarm panels". Several questions are bound to this approach, but this place is not to discuss them.

Thank you for being open to cooperate !