HASwitchPlate/HASPone #147: Exception when trying to uncheck "mDNS enabled" and "Save Settings"

rebel7580, 2021-11-11:
I have 2 hasps, both with "mDNS enabled" checked. I upgraded to 1.03 successfully. Later I decided that mDNS was not needed, so unchecked it, and did "Save Settings". 

Found that when the web page returned (see note 2 below,) mDNS was still checked and an exception was shown. Both hasps had the same exception details.

HASP Version: 1.03
LCD Model: NX3224T024_011R
LCD Version: 3
LCD Active Page: 1
LCD Serial Speed: 921600
CPU Frequency: 160MHz
Sketch Size: 658832 bytes
Free Sketch Space: 2486272 bytes
Heap Free: 17288
Heap Fragmentation: 3
ESP core version: 3.0.2
IP Address: 192.168.123.25
Signal Strength: -48
Uptime: 15
Last reset: Fatal exception:28 flag:2 (Exception) epc1:0x4023f274 epc2:0x00000000 epc3:0x00000000 excvaddr:0x122c0000 depc:0x00000000

Note 1: Since I didn't try this before I upgraded, it's not clear this has anything to do with the new version.
Note 2: The first hasp I did this on returned to the main web page after a few seconds. The second one never did. Had to re-access it via a refreshed IP address.

aderusha, 2021-11-15:
Turns out to be a race condition in the config save process, working on pushing an emergency update with the fix here shortly.

aderusha, 2021-11-15:
dev build of fix available [here](https://raw.githubusercontent.com/HASwitchPlate/HASPone/1.04-HASPone-firmware/Arduino_Sketch/HASwitchPlate.ino.d1_mini.bin) for testing:

`https://raw.githubusercontent.com/HASwitchPlate/HASPone/1.04-HASPone-firmware/Arduino_Sketch/HASwitchPlate.ino.d1_mini.bin`

rebel7580, 2021-11-15:
Dev build seems to work. Was able to uncheck "mDNS enabled" and save without issues.
Thanks!

aderusha, 2021-11-18:
Version 1.04 pushed to `master` with fix for this issue.

closed 2021-11-18 by aderusha