HASwitchPlate/HASPone #148: Generalize GPIO access on D2

aderusha, 2021-11-11:
Currently the HASPone project allows activating GPIO D2 in order to control a beeper.  Some of this interaction is hard-coded to respond to screen presses, it might be more useful if this was generalized to allow for wider use cases.

rebel7580, 2021-11-11:
Good idea! How were you thinking this would work? I'm always interested in exploiting this kind of thing. :)
On one of my hasps, I use D1 for a  motion sensor, which I mounted on a modified 3 gang wall plate (your 3d design).

EDSTOBI, 2021-11-13:
hello
I use HASPone with node red, because I use it to control my pond. I have the wish to send a mqtt command from a source, no matter if from homeautomation, node red or any esp to activate the peep on hasp/plate. I am totally flexible if I send the signal to hasp/plate02/alarm/on or any other json mqtt command.
I would suggest that these commands are structured like the control commands for Nextion display on or brightness.