HASwitchPlate/HASPone #173: Stuck on wifi connected

scubieman, 2022-01-04:
I use to have the old haswitchplate installed on my HA. Now trying to upgrade to HASPone.

My device keeps getting stuck on WIFI connected. I tried factory resetting. Double checking names. And nothing is working. I can see if it Blueprint as device. Any ideas what issue is?

aderusha, 2022-01-04:
OK so first, make sure you [go through this process to clear out the old HASP stuff](https://github.com/HASwitchPlate/HASPone/wiki/Updating-legacy-HASP-to-HASPone).  If you're still not connecting (or things just aren't working out right for you in general), [grab a debug log](https://github.com/HASwitchPlate/HASPone/wiki/Capture-HASPone-debug-output) and post it here, then we can work through what is happening on your device and hopefully get you moving.

scubieman, 2022-01-04:
How would i delete switch plates added to HA accidently? Or Change the name? Example if i had plate01, and I switched name to brplate. It will show plate01 and brplate in list.

aderusha, 2022-01-04:
In Home Assistant, go to Configuration > Devices & Services > Devices tab > find the old device (plate01) and click on it > Delete.

scubieman, 2022-01-04:
okay they connect to wifi , but stuck on "wifi connected" and dont do anything useful

aderusha, 2022-01-04:
Allright, can you post a debug dump of the whole process?  Is this device currently in the wall?

scubieman, 2022-01-04:
its sitting next to me. I have two of them, I can not access the ip address. Generally what works is unplugging a bunch of times, but i still have some issues
want to chat on DRzz discord?

scubieman, 2022-01-05:
@aderusha So device says wifi connecting..... However I show it active in HA now. I tried running the automation HASPone Core functionality on both devices but fails to do anything
[image]
.

scubieman, 2022-01-05:
Lowered the serial speed down to 115200. It appears to be working now.

closed 2022-01-05 by scubieman