HASwitchPlate/HASPone #187: hasp_Display_Text.yaml fails on import

aderusha, 2022-05-06:
### Discussed in https://github.com/HASwitchPlate/HASPone/discussions/186

<div type='discussions-op-text'>

<sup>Originally posted by **robellis-zz** May  5, 2022</sup>
When i try to import the hasp_Display_Text.yaml blueprint i get an 'unknown error' is there a different way to import or can i copy and paste it into HA?</div>

aderusha, 2022-05-06:
The scope of this is much wider than this one blueprint, issue seems to have cropped up in 2022.05 and impacts anything using a slider-type number select.

aderusha, 2022-05-06:
@robellis-zz Question for you - what version of Home Assistant are you currently running where you are seeing this import error?  2022.05 or some older version?

aderusha, 2022-05-06:
Apparently this is a bug in 2022.05: https://github.com/home-assistant/core/issues/71396

robellis-zz, 2022-05-06:
Makes sense. I am running 2022.5.1 right now

aderusha, 2022-05-06:
Looks like a fix has been merged, so we should see resolution in the .2 point release.

aderusha, 2022-05-07:
Allright, the 2022.5.2 release has fixed the issue reported above.  However, there are reports of other problems that are related.  I don't think the remaining issues should impact anything we're doing with the HASPone blueprints, but I'm going to leave this open until I see that those additional problems have been patched just to make sure.

aderusha, 2022-05-08:
Looks like we should be all set, the release version of  HA (2022.05.2) fixes all HASPone related issues and a patch is on the way for the rest of the issues (which don't impact HASPone).  Closing this issue, I think we should be set!

Thanks again to @robellis-zz for the report and helping troubleshoot.

closed 2022-05-08 by aderusha