HASwitchPlate/HASPone #188: Any way to reboot the ESP from the touchscreen?

mattster98, 2022-05-16:
I had a failure mode recently I haven't encountered before where the touch panel is working, but it is not on the network or connected to MQTT.  I eventually had to disconnect power to recover.  It came back fine - not sure what the issue was.

But, it made me realize I don't think there's any other way to force a reboot other than via MQTT.  I saw an option in the code to force a reboot when p[0].b[1] is pressed, but it's false by default.  Is something like that the best/only option?  I see you can restart the nextion lcd with a button command, but I don't think that would have helped.

I was thinking long-pressing any button for more than X seconds might be a reasonable watchdog-style reboot mechanism.  Not sure if that's possible.  Or, if the ESP can see the nextion reboot I could use that option to trigger it.

aderusha, 2022-05-16:
`rebootOnp0b1` is set `true` while the code is handling the [WiFi connection](https://github.com/HASwitchPlate/HASPone/blob/main/Arduino_Sketch/HASwitchPlate/HASwitchPlate.ino#L1962) or [MQTT connection](https://github.com/HASwitchPlate/HASPone/blob/main/Arduino_Sketch/HASwitchPlate/HASwitchPlate.ino#L326) with the intent of giving the user a chance to reboot the system during these steps if something isn't working.  

A long press (or maybe "press 5 times in two seconds" or similar might be another solution that could be setup to run at all times.  I'll add this as a feature request and maybe figure something out that works a little better than the current solution.

aderusha, 2022-05-16:
Allright, added this feature to the current 1.06 development branch.  Hold any button down for 10 seconds (or more), release, and the device will trigger a reboot.