HASwitchPlate/HASPone #196: Alarm panel not working (Page 7)

Stephan1908, 2022-07-19:
Hello,
i can't activate an alarm in the alarm viewer. If I enter the code and then press arm away or arm home, nothing happens. What am I doing wrong?
My blueprint looks like this:

alias: HASPone p[7].b[all] displays an alarm control panel
description: ''
use_blueprint:
  path: HASwitchPlate/hasp_Display_Alarm_Control_page7_new.yaml
  input:
    b15armservice: alarm_arm_away
    haspdevice: be9fd115990418ac490f4dbd6727839e
    alarmpanel: alarm_control_panel.alarm
    b13armservice: alarm_arm_home

Thanks
Stephan

bsmith76s, 2022-12-11:
I am having the same issue. My blueprint yaml

```

alias: HASPone Plate 2_Alarm_all  displays an alarm control panel
description: ""
use_blueprint:
  path: HASwitchPlate/hasp_Display_Alarm_Control_page7.yaml
  input:
    haspdevice: 9adc68c9e7a5b6aa48de3c3ddc0be4f0
    alarmpanel: alarm_control_panel.ha_alarm
    b13armservice: alarm_arm_away
    b15armservice: alarm_arm_home
```

austwhite, 2023-06-18:
I can't replicate this issue.   I am running it with Alarmo (fro HACS) and it works flawlessly for me.   
Arm Home and Arm Away both work as does disarm function.......