HASwitchPlate/HASPone #198: Errors with Blueprints

omayhemo, 2022-07-22:
After my system was shut down for some time, all the HASPone I have in the house, blanked their display back to the basic core configuration.  I've attempted running the BPs again, even going back and running the core configuration BP, but I get errors whenever I run them.  I also get the following error on HA startup.  So all my HASPone are currently not programmable.  I'm on the latest Blueprint Version: 1.05.00 for everything.

The following error on HA startup:
```
2022-07-22 08:13:32 ERROR (MainThread) [homeassistant.components.automation] Blueprint HASPone p[x].b[y] displays a toggle button generated invalid automation with inputs OrderedDict([('ycen_select', '1 - Centered'), ('haspdevice', '6d4c8c9161af143210a835cdf66845ee'), ('entity', 'input_boolean.master_bedroom_automation'), ('state_on', 'on'), ('text_on', '||'), ('text_off', '>'), ('on_fgcolor', 0), ('off_bgcolor', 0), ('off_fgcolor', 0), ('action_on', [OrderedDict([('service', '')])]), ('action_off', [OrderedDict([('service', '')])]), ('xcen_select', '1 - Centered'), ('hasppage', 10), ('font_select', '8 - Noto Sans 64'), ('haspbutton', 7), ('on_bgcolor', 0)]): Service  does not match format <domain>.<name> for dictionary value @ data['action'][0]['choose'][2]['sequence'][0]['choose'][0]['sequence'][0]['choose'][0]['sequence'][0]['service']. Got None
```

The following error when I run any BP and repeatedly/intermittently:
```
2022-07-22 08:21:23 ERROR (MainThread) [homeassistant.components.automation.hasp_master_p10b1_displays_a_toggle_button] Error rendering variables: UndefinedError: None has no element 0
```

aderusha, 2022-07-22:
It looks like you don't have anything entered for the "action on" part of the blueprint.  If you go into the editor, I suspect you'll find that "Service" has been selected but no actual service call configured.  It's failing trying to execute that empty service call.

omayhemo, 2022-07-22:
The action is filled out on all the BP automations.
Example:
[image]

omayhemo, 2022-07-22:
Built a new automation from BP from scratch just to be sure (just to be clear, nothing has changed in these automations since they were working that I'm aware of).  Still the same error.  Both off and on actions were fully filled out.

Simplified with a group, didn't help.
[image]

[image]

aderusha, 2022-07-22:
Can you post the trace from running the automation?

omayhemo, 2022-07-22:
Let me know if this is what you're looking for.  HASPone is the only thing I use HA automations for (prefer NetDaemon), so not real familiar with debugging them.

```
{
  "trace": {
    "last_step": null,
    "run_id": "7b605e4157c5c148efb105e611ac3100",
    "state": "stopped",
    "script_execution": null,
    "timestamp": {
      "start": "2022-07-22T15:02:03.296829+00:00",
      "finish": "2022-07-22T15:02:03.305029+00:00"
    },
    "domain": "automation",
    "item_id": "1658500814791",
    "error": "UndefinedError: None has no element 0",
    "trigger": null,
    "trace": {},
    "config": {
      "mode": "parallel",
      "max_exceeded": "silent",
      "variables": {
        "haspdevice": "6d4c8c9161af143210a835cdf66845ee",
        "haspname": "{%- for entity in device_entities(haspdevice) -%}\n  {%- if entity|regex_search(\"^sensor\\..+_sensor(?:_\\d+|)$\") -%}\n    {{- entity|regex_replace(find=\"^sensor\\.\", replace=\"\", ignorecase=true)|regex_replace(find=\"_sensor(?:_\\d+|)$\", replace=\"\", ignorecase=true) -}}\n  {%- endif -%}\n{%- endfor -%}",
        "hasppage": 10,
        "haspbutton": 8,
        "entity": "light.master_lamps_2",
        "attribute": "state",
        "state_on": "on",
        "action_on": [
          {
            "service": "light.turn_on",
            "data": {},
            "target": {
              "entity_id": "light.master_lamps_2"
            }
          }
        ],
        "action_off": [
          {
            "service": "light.turn_off",
            "data": {},
            "target": {
              "entity_id": "light.master_lamps_2"
            }
          }
        ],
        "text_on": "Turn Off",
        "text_off": "Turn On",
        "on_fgcolor": -1,
        "on_bgcolor": -1,
        "off_fgcolor": -1,
        "off_bgcolor": -1,
        "font_select": "8 - Noto Sans 64",
        "font": "{{ font_select.split(\" - \")[0] | int }}",
        "xcen_select": "1 - Centered",
        "xcen": "{{ xcen_select.split(\" - \")[0] | int }}",
        "ycen_select": "1 - Centered",
        "ycen": "{{ ycen_select.split(\" - \")[0] | int }}",
        "wrap": false,
        "haspobject": "{{ \"p[\" ~ hasppage ~ \"].b[\" ~ haspbutton ~ \"]\" }}",
        "commandtopic": "{{ \"hasp/\" ~ haspname ~ \"/command/\" ~ haspobject }}",
        "jsontopic": "{{ \"hasp/\" ~ haspname ~ \"/state/json\" }}",
        "jsoncommandtopic": "{{ \"hasp/\" ~ haspname ~ \"/command/json\" }}",
        "isbr": "{% if wrap == true %}1{% else %}0{% endif %}",
        "selectedfgtopic": "{{ \"hasp/\" ~ haspname ~ \"/light/selectedforegroundcolor/rgb\" }}",
        "selectedbgtopic": "{{ \"hasp/\" ~ haspname ~ \"/light/selectedbackgroundcolor/rgb\" }}",
        "unselectedfgtopic": "{{ \"hasp/\" ~ haspname ~ \"/light/unselectedforegroundcolor/rgb\" }}",
        "unselectedbgtopic": "{{ \"hasp/\" ~ haspname ~ \"/light/unselectedbackgroundcolor/rgb\" }}",
        "selectedfg": "{%- set color = namespace() -%} {%- for entity in device_entities(haspdevice) -%}\n  {%- if entity|regex_search(\"^light\\..*_selected_foreground_color(?:_\\d+|)$\") -%}\n    {%- set color.source=entity -%}\n  {%- endif -%}\n{%- endfor -%} {%- set brightness = state_attr(color.source, \"brightness\")|int(default=255) / 255 -%} {%- set red=(state_attr(color.source, \"rgb_color\")[0] * brightness)|int(default=0) -%} {%- set green=(state_attr(color.source, \"rgb_color\")[1] * brightness)|int(default=0) -%} {%- set blue=(state_attr(color.source, \"rgb_color\")[2] * brightness)|int(default=0) -%} {{ (red|bitwise_and(248)*256) + (green|bitwise_and(252)*8) + (blue|bitwise_and(248)/8)|int }}",
[... 260 lines cut ...]
                "entity_id": "light.master_lamps_2"
              }
            }
          ],
          "action_off": [
            {
              "service": "light.turn_off",
              "data": {},
              "target": {
                "entity_id": "light.master_lamps_2"
              }
            }
          ]
        }
      }
    },
    "context": {
      "id": "01G8K6W3S04EW52BVSA3P947D7",
      "parent_id": "01G8K6W331BB32797NJGYH6D94",
      "user_id": null
    }
  },
  "logbookEntries": []
}
```

aderusha, 2022-07-22:
Huh.  I'm trying to piece these together as we're looking at logs from several different automations.  Taking the most recent post with the full log "Master HASPone - Lamps", I notice that the path is `HASwitchPlate/hasp_Display_Toggle 2.yaml` and I'm curious about the ` 2` part there.  Would it be possible to clean house (nuke automations, clear out `blueprints/HASwitchPlate` and re-import) or would that be a huge undertaking?

Also, can you show me what Home Assistant logs in the main system log when you fire off the "Master HASPone - Lamps" automation?

omayhemo, 2022-07-22:
I have no idea what that path is from, other than I create directly from BP and then rename and configure.  I didn't mess around with the yaml.

While not optimal, I will try clearing everything and starting from scratch.  It's not trivial, but it's not huge either.

closed 2022-08-16 by omayhemo