HASwitchPlate/HASPone #199: Temperature blueprint - doesn't save

SteveRMann, 2022-07-23:
First noticed when I keep turning this option off, but it won't save.  In fact, turning the option off does not prompt for "save". I have to change something else to get the save button.  But it doesn't save the automation.
[image]

Further, I deleted the automation and the blueprint, then downloaded the blueprint again, clicked on create automation, clicked on save. And there was no "run automations" at the top of the blueprint.  And there is no automation saved in "Automations".

Tried is on a different device. Create the temperature button, click on save then run actions. Works fine. Open the automation and turn on the colorize option, click "save" then - nothing.  The "run automations" button did not appear and the saved automation

SteveRMann, 2022-07-23:
Update-
I opened automations.yaml and found three copies of the automation:

`- id: '1658588868806'
  alias: desktop - HASP current temperature
  description: ''
  use_blueprint:
    path: HASwitchPlate/hasp_Display_Weather_Temperature_with_Icon_and_Colors.yaml
    input:
      haspdevice: abdbcdf0fb55720ab6b413e4fe8f7720
      suffix_select: °F
- id: '1658589448746'
  alias: desktop - HASP Current Temperature
  description: ''
  use_blueprint:
    path: HASwitchPlate/hasp_Display_Weather_Temperature_with_Icon_and_Colors.yaml
    input:
      haspdevice: abdbcdf0fb55720ab6b413e4fe8f7720
      suffix_select: °F
      roundtemp: true
- id: '1658589610897'
  alias: desktop - HASP Current Temperature
  description: ''
  use_blueprint:
    path: HASwitchPlate/hasp_Display_Weather_Temperature_with_Icon_and_Colors.yaml
    input:
      haspdevice: abdbcdf0fb55720ab6b413e4fe8f7720
      haspbutton: 6
      weather_provider: weather.openweathermap
      suffix_select: °F
      roundtemp: true
`

That doesn't seem right.

aderusha, 2022-07-23:
You are very likely running into [this bug in Home Assistant](https://github.com/home-assistant/frontend/issues/13206).  Responding to that bug or adding a "thumbs up" might help get some visibility on the issue.