HASwitchPlate/HASPone #200: Won't Stay Connected

omayhemo, 2022-08-21:
This is the first HASPone I installed.  It's no longer staying connected and constantly reboots to the QR code.  Most of the time sits on connecting to Wi-Fi.  I've tried setting it up with the BPs again, but it's hard to keep online, usually crashes and heads back to the Wi-Fi connecting screen or QR code.

I've attempted to follow these instructions https://github.com/HASwitchPlate/HASPone/wiki/Flashing-HASPone-to-an-ESP8266
that I found recommended for a similar now closed issue, but when I plug it into the computer it never registers a port, the blue light on the ESP lights briefly then goes out and no further activity.  Seems like it maybe bricked.

omayhemo, 2022-08-27:
Still won't connect to the wifi, I'm assuming the ESP wifi is bricked.

aderusha, 2022-08-27:
Any chance you have a debug log to share to see if the device is trying to tell us something?  Does moving the device physically result in a different outcome?  Trying to figure out if we're seeing a signal issue or code issue or a failed ESP8266.  I've yet to see just the WiFi fail, but hey there's a first time for everything maybe?

omayhemo, 2022-08-27:
I have 4 UniFi APs in my home.  One of them is within 15' of the HASP, I've tuned my WiFi by measuring every corner of my home because I'm obsessive like that, signal strength is definitely not a problem.  You may (or probably not) remember I recently had a problem with a rogue automation that was preventing me from using the BPs to program things.  You suggested I reset.  I did that.  That's when it started dropping offline intermittently.  So I flashed it.

I can't get the HASP online at all anymore, since I flashed it, so I don't believe I can provide a debug log.  At this time, it's sitting at the QR code (and has been since flash), and I cannot connect to it's internal AP, so I can't do anything with it.

aderusha, 2022-08-27:
If the device is showing the QR code then it's alive in some sense and should be spitting out debug logs via serial while doing so.  I don't want to have you running in too many circles here, so if you'd like to setup a return, hit me up at luma@svideo.org and we can get that worked out that way.

Otherwise, getting a serial debug log will require disassembling the device.  Another alternative is to try out the latest dev build of 1.06 which has several fixes around the wifi stack that has improved connection performance for my devices here where I also have 4 Unifi APs.  That can be flashed using the process you linked above and you can find that build here: https://github.com/HASwitchPlate/HASPone/raw/dev-1.06/Arduino_Sketch/HASwitchPlate.ino.d1_mini.bin

omayhemo, 2022-08-31:
I'll try flashing 1.06 and see if I can get the logs too, will be a useful exercise for me.

ssurovich, 2022-09-06:
I flashed 1.06 and it just sends my panel into an endless loop of Connecting to WiFi and you can see the initial screen with my buttons but then it reboots - endless loop of this.