HASwitchPlate/HASPone #201: Getting an error when using NodeMCU-PyFlasher - File Version is too low

rkhanso, 2022-08-29:
Getting an error when using NodeMCU-PyFlasher - File Version is too low
Is this a problem with the HASwitchPlate.tft  that I downloaded? Something else?

rkhanso, 2022-08-29:
Someone from the Nextion forum said the following. 

Apparently, the firmware of your Nextion is more recent than the editor version used to compile the tft file. That’s not a SD card problem. That’s because the one who created the tft file did compile it a long time ago and/or using a now outdated editor version.

If I were you, I’d ask the guy to simply recompile it from his .hmi source with the current editor 1.63.1 for you and other users who purchased their screens recently. Or I’d download and install the current Nextion Editor myself, load the .hmi file from GitHub and recompile it.

aderusha, 2022-08-29:
The HASPone project has two firmware images, one for the ESP8266 and one for the Nextion display.  The program `NodeMCU-PyFlasher` can be used to flash an ESP8266 with the ESP8266 firmware file found [here](https://github.com/HASwitchPlate/HASPone/blob/main/Arduino_Sketch/HASwitchPlate.ino.d1_mini.bin).  

`HASwitchPlate.tft` is the Nextion firmware, which you can apply to the Nextion through the HASPone web interface of via an SDcard.  If you want to open the Nextion project in the Nextion editor, the file [HASwitchPlate.HMI](https://github.com/HASwitchPlate/HASPone/blob/main/Nextion_HMI/HASwitchPlate.HMI) is the source document for this project and can be edited as needed.

Make sense?  It seems your core problem is attempting to load a Nextion firmware into an ESP8266 firmware flashing tool, which isn't going to work out.

rkhanso, 2022-08-30:
I'm pretty sure I'm trying this correctly. I'm not trying to put the ESP8266 firmware on the Nextion display.  I see where I made a mistake in my intitial comment and title. That was definitely a mistake. It should NOT have said using NodeMCU-PyFlasher. I'm sorry for the confusion.  I'll see if I can change it.

I have an 8GB Micro SD Card that I formatted with FAT-32.  I copied HASwitchPlate.tft onto the SD Card and inserted it into the Nextion Display's slot and then booted it up. I get the error about the File Version being too low.

aderusha, 2022-08-30:
Huh.  That is pretty unexpected, I have a batch of Nextion devices that I received last week from ITEAD that are all flashing with the current revision, that was generated on 1.61.  1.63 has some changes that result in even less available memory on the device and, as a result, the current project cannot compile.  It is possible to go and make some changes to the project to remove some fields etc to free up additional memory, but you'd be running a version of the firmware that is slightly less capable than the code that's in the project.

Here's a thought - can you launch the upgrade from the HASPone web interface and see what it does?  The good news with Nextion is that they're nearly impossible to brick, so trying a firmware flash that goes sideways shouldn't leave you in a state that you cannot recover from.

rkhanso, 2022-08-30:
I guess some people have had trouble with certain SD Cards. I use the generic MicroCenter SD Cards. Maybe I need to get name-brand instead.

Question about the pins to connect the wires between Nextion and 8266. Blue and Yellow wires go to which Pins on the D1 Mini/8266?   I've seen references to the Tx and Rx pins on the 8266, and also D4 and D7.

rkhanso, 2022-08-30:
Nevermind. I got it connected. D7 and D4.
But I'm not sure how to do an upgrade/update to get the Nextion file on it. Won't I just be updating the ESP/HASPone firmware?

rkhanso, 2022-08-30:
Boy - I feel stupid. I just messed around a little more and figured out what you were talking about. 

But I get the same error:   File Version is too low

So I guess it's probably not the SD Card that's the problem then.

rkhanso, 2022-08-30:
The Nextion display I just received from Amazon was from your link on Github. It has NX3224T024_011 on the back. There's a sticker over the SD Card Slot that says DE607C43435D5236. On the edge of the display it has HST024042 3-2.

Since I'm getting the "File Version is too low" when trying both the SDCard and ESP8266 method of installing the HASwitchPlate.tft file, do you think I'm doing everything correct and there is some other issue that is going on? Something that can be fixed? Or should I return the display to Amazon?  I'd really like to get this HASPone working since I've spent this money on it.

rkhanso, 2022-08-30:
I think I got it. I hope I didn't mess anything up though....
The Nextion Editor seemed to indicate to me that the only problem was p0.
That appears to be a QR code to use with your phone to set up WiFi with the HASPone. There was a gray rectangle in front of the QR code. I clicked it and hit delete. I then compiled, put that file on an SD Card and uploaded it to the display. It sayd Update Successful! 

Powered it up and it connected to WiFi. MQTT is connected.  I think I just need to go through the Home Assistant setup now, right?

aderusha, 2022-08-30:
The editor will fail to compile because the project runs out of memory after recent updates which reduce the overall memory available.  It told you about p0 because that's the first page, but in general, one just needs to remove something from the overall project.  You've removed the QR code which means you won't see the code during the initial setup process, but otherwise everything should work as normal.

Now run through the setup!

djqpaz, 2022-09-04:
[rkhanso](https://github.com/rkhanso) any update? i just recently kicked off this as a project.  Seem as though there is a problem with the incompatibility with newer firmware and older tft files for the screen.  

I ordered a USB to TTL to try and touble shoot, but not sure what to do next.

rkhanso, 2022-09-05:
I got it to work by removing a gray rectangle on the p0 page. After I did that, I could install the firmware since the size was a little bit smaller. The only thing that affects is showing the p0 page (QR code) - but the display still shows the IP address/SSID....so no trouble in the end. You don't really need the p0 page to get it set up.

A little more info - in the Nextion Editor, load the .hmi.  On page p0, click on the gray rectangle over the QR code area and delete it. Then you can compile successfully and put it on the the SD Card and update the display like normal using the SD Card method.

aderusha, 2022-09-05:
I've just updated the main project branch with a new HMI and TFT file for the Nextion Basic to address this issue.  Removing components will get the job done but will make your project incompatible with some blueprints. 

As no new features are added and this only impacts the Basic model, I've not incremented the version in the project as I don't want people being lit up with an update notification.  So, LCD version stays at 3, but if you are having problems getting it loaded on a new Nextion device, try re-downloading the link from the repo and trying again (via web upload, SD card, etc).

closed 2022-11-10 by aderusha