HASwitchPlate/HASPone #213: Dimmer slider issues

Gurulu-pc, 2022-11-19:
Might be my configurations but when I slide the dimmer in home assistant, my HASPone slider moves accordingly and the light obeys. When i move the slider on the HASPone, nothing happens. So basically, the HASPone slider is only a status indicator instead of a control device. The dimmer is a DIY lamp I've built with esphome and a led strip. It's very strait forward PWM dimming from a wemo D1 mini. The dimmer was configured from the default blueprint from this repo (hasp_Display_Dimmer_with_Icon.yaml)
Software version : Home Assistant 2022.11.3, Supervisor 2022.10.2, Operating System 9.3
hardware: RPI3B+
here's the relevant automation.yaml entry :
`- id: '1667772236417'
  alias: HASPone p[x].b[y] displays a dimmer with a toggle on/off icon
  description: ''
  use_blueprint:
    path: HASwitchPlate/hasp_Display_Dimmer_with_Icon.yaml
    input:
      font_select: 6 - Noto Sans 32
      haspdevice: 43f87b51c35fe9a31c395bb715839774
      haspbutton: 8
      dimmer: light.salle_de_bain
      text_on: Lum Pharmacie |
      ycen_select: 1 - Centered
      icon_off: 
      icon_on: 
      xcen_select: 2 - Right aligned
      hasppage: 4
`
Thanks for your help

aderusha, 2022-11-20:
What happens when you move the slider on the HASPone?  Does the slider move and stay where you dragged it on the display?

If you look at device `light.salle_de_bain` in Home Assistant when you do this, does the brightness change in HA at all?  What happens if you change the brightness in HA, do you see any change on the slider on your HASPone?

Gurulu-pc, 2022-11-20:
> What happens when you move the slider on the HASPone? Does the slider move and stay where you dragged it on the display?

Yes, slider moves and stay where it was dragged. 
> If you look at device `light.salle_de_bain` in Home Assistant when you do this, does the brightness change in HA at all?

No, nothing changes on HA's side. 

> What happens if you change the brightness in HA, do you see any change on the slider on your HASPone?

Yes, HASPone slider moves accordingly and stay in the position of the new brightness value.

Also, my mqtt server is hosted by another machine. I don't know if this is relevant to the issue. Doesn't seem like it should be a problem at all but could be. I'm not running firewall on my local network.

aderusha, 2022-11-20:
Do you get anything in the automation log when you drag the slider?

Gurulu-pc, 2022-11-24:
Nothing happens in the logs and mqtt server is not receiving anything. I did tweak the nexion interface to my liking. I'll compare the default tft file from the repo and mine just to be sure.

closed 2026-03-28 by aderusha