HASwitchPlate/HASPone #227: Device hangs with 1.05

SylvainGa, 2023-02-28:
When my devices were using HASP 0.40 for over a year, they were rock solid. Never had a device hang. Now that I've upgraded to 1.05, all of them (I have four) have at least hung once and that is just in a few months. The screen is unresponsive (well, it register the press but that's it) and its web interface isn't working either. It takes a breaker reset to bring them back. What should I try to try to troubleshoot what's going on? Thanks.

SylvainGa, 2023-03-07:
Thanks for the reply. Since that message, one of the frozen screen changed from frozen at the 'Scenes' page to frozen at "Connected to MQTT / Connected to Wifi" for a few days and yesterday, it's responsive again. That's the device I wanted to disconnect and plug it through USB.

One question, on Windows, I can't get the logs at boot since when I connect the USB cable, the serial terminal I'm using (Termite) do not know about the virtual serial port until it's connected and by the time I activate it, the boot data is long gone. Do you know of a serial terminal program that doesn't have that limitation or a way to hard reboot a hung Wemos D1 while maintaining its serial port connection?

aderusha, 2023-03-07:
I do!  I've been using [SimplySerial](https://github.com/fasteddy516/SimplySerial) which is a window terminal app that will handle disconnect/reconnect and can usually pick up boot messages.

aderusha, 2023-10-03:
One common situation we've encountered are problems with the WiFi.  @jwhamlin244 would it be possible to try and reboot the WiFi AP that your HASPone is attached to and see how it responds?

jwhamlin244, 2023-10-05:
@aderusha I have turned off my router -> no change on haspOne -> restarted router -> still no change on haspOne. 
I have not noticed any internet issues, although i can't rule out that it has never blipped out without me noticing at some point.