HASwitchPlate/HASPone #234: Blueprint for Heatpump (both Pge6 and Pge9) are not working HA 2023.6.x

austwhite, 2023-06-18:
Hi,
Just wanted to let you know the blueprint for the Heatpump control, both Page 6 and Page 9 versions fail.
The automation will load initially and the page display.   I can then press Heat and change the temperature, but when I hit Off it completely freezes up and never runs the automation again.   
This happens with both the Page 6 and Page 9 versions.   

This is not a critical thing, as I can use different options to control HVAC systems, it would just be a nice to have on the HASPOne :)

Thanks :)

austwhite, 2023-06-21:
As an additional comment.  The issue seems to happen when "off" is selected and the automation doesn't run when the HVAC is in "off" mode" .  If I set the HVAC to Heat or Cool mode, then the automation resumes running and I can control everything from the HASPOne.   As soon as the HVAC is switched to OFF, the automation stops running again.
Happens with both Pge6 and Pge9 versions, so looks to be something when climate.xxxx is in the "off" mode.

BenJewell, 2024-08-26:
Did you ever find a solution for this?

austwhite, 2024-10-26:
No soloution found for this.  Will have to wait for th developer to look at it I think.

austwhite, 2024-12-29:
Looks like these work fine with the Generic Thermostate now, though there is no way to tell the blueprint it is heat only or cool only, so it still displays the heat cool button.    
Still completely not working with Z-Wave thermostat, so much so that the automation doesn't even run when a Z-Wave thermostat is selected.

closed 2025-07-26 by austwhite