HASwitchPlate/HASPone #236: Mqtt issue with new HA 2023.8.0

farberm, 2023-08-03:
Some MQTT entities have an entity name that starts with the device name. This is not expected. To avoid a duplicate name the device name prefix is stripped of the entity name as a work-a-round. Please inform the maintainer of the software application that supplies the affected entities to fix this issue.

List of affected entities:

light.plate01_backlight
light.front_porch_light_button_c
number.plate01_active_page
light.front_porch_light_button_d
light.plate01_selected_foreground_color
light.plate01_unselected_foreground_color
sensor.plate01_sensor
light.plate01_selected_background_color
light.front_porch_light_button_b
light.plate01_unselected_background_color
light.front_porch_light_button_a

aderusha, 2023-08-03:
Sounds like a new breaking change in Home Assistant which conveniently isn't mentioned in the breaking changes.  Sounds like we have 6 months to address this and the errors can be ignored for the moment.  I'm digging into this and will get a solution ready ASAP.

Forum thread on the topic here: https://community.home-assistant.io/t/psa-mqtt-name-changes-in-2023-8

farberm, 2023-08-03:
ThxSent from my iPad

austwhite, 2023-08-04:
This is a new change to clean up MQTT entities in Home Assistant.
Same error here, and I also get errors on my ESPPresense devices (different project of course).  

Looks like all the blueprints and MQTT entities will need to be updated to match the new MQTT naming conventions

lebherz, 2023-09-28:
hey @aderusha 
any idea to fix it? or should we leave this project 👎

aderusha, 2023-09-28:
A firmware update is being worked on for this and a couple other changes.  It will also require a blueprint update.  You should no longer be seeing notifications on the problem after the HA devs backed off on the panic button.  We have 6 months so I'm going to use this time to see if a few more things can't make their way into the change too.

So no need to worry yet, I have these things all over my house and use them constantly and so I'm going to keep this going. My house would be a lot dumber without HASPone.

aderusha, 2024-07-24:
1.06 addresses this issue with the new MQTT discovery format.

closed 2024-07-24 by aderusha