HASwitchPlate/HASPone #242: Stuck at IP Address Screen

omayhemo, 2024-04-22:
Noticed a few days ago my configuration had blanked (all the buttons were now blank).  Try as I might, I've been unable to get any configuration to take.  So I reset the device and started from scratch.  Now I'm stuck at the IP address screen.  Both the WiFi and the MQTT addresses are set and working.  The odd thing, I can actually use the entities to for example turn the backlight off on the screen.  So I know the device is talking with HomeAssistant.  However, none of the config blueprints are having any affect.

aderusha, 2024-04-22:
Can you try running through some of the solutions shown here and let us know how it goes? https://github.com/HASwitchPlate/HASPone/wiki/Fix-common-HASPone-issues

omayhemo, 2024-04-24:
No joy.  I removed it completely from HA, reset it, set it up from scratch.  I can move it from page to page using the page number property, but the blueprints are not configuring those pages anymore.

aderusha, 2024-04-24:
Anything showing up in your HA logs?  Another idea, I have a beta release that's probably going to be an RC unless someone finds additional bugs which you can try out here: https://raw.githubusercontent.com/HASwitchPlate/HASPone/dev/Arduino_Sketch/HASwitchPlate.ino.d1_mini.bin

aderusha, 2024-04-24:
How about a new automation deployed from blueprint?  eg, run the Clock or whatever blueprint on an unused button, does that show?

omayhemo, 2024-04-24:
Assuming I understood correctly, I've added a clock as button 7 on page 1.  Still nothing showing.  It maybe worth noting, that while I can change the page with the page number property (I can witness the page changing) and I can toggle the backlight, I cannot get the foreground and background (really just background because none of the buttons are displaying a foreground item) color changes to take.

omayhemo, 2024-05-08:
Haven't been able to fix it, and haven't had time to focus on it.  Will try to you new bin.

WebDesignerlol, 2024-05-11:
> Checking in - is this still not working at all for you? I'm getting the last pieces together to release 1.06, and I've pushed some more changes to: https://raw.githubusercontent.com/HASwitchPlate/HASPone/dev/Arduino_Sketch/HASwitchPlate.ino.d1_mini.bin
> 
> Just for test, can do you:
> 
> * Flash 1.06 from the link above
> * Rename the HASPone to some test value
> * Deploy a new core functionality blueprint for the test name

I am not the original poster of this issue but i am experiencing all the same issues. I attempted to try your 1.06 firmware to no avail either. Maybe the blueprints need to be updated for the 1.06?

aderusha, 2024-05-12:
> I am not the original poster of this issue but i am experiencing all the same issues. I attempted to try your 1.06 firmware to no avail either. Maybe the blueprints need to be updated for the 1.06?

Did you deploy a new Core Functionality blueprint or are you using the one you already had?  Trying to get a handle on the error situation here.

WebDesignerlol, 2024-05-12:
> > I am not the original poster of this issue but i am experiencing all the same issues. I attempted to try your 1.06 firmware to no avail either. Maybe the blueprints need to be updated for the 1.06?
> 
> Did you deploy a new automation from the Core Functionality blueprint or are you using the one you already had? Trying to get a handle on the error situation here. There are a handful of changes to some blueprints in 1.06 but nothing substantial in the core.
> 
> edit: I just wrapped up the release of 1.06 and, despite my lies just moments ago, it might have some useful bits in the Core Functionality blueprint. Try updating that as well!

Sorry I was exhausted after working on it for hours yesterday. I ended up trying nearly everything. I did an update to the D1 mini, both from URL and manual download, same for the TFT (which took me a while to figure out, maybe some update of the documentation regarding the screen types might help - for example i found out that the nextion display I had in mine did not have an "R" in the model number and that if it has a "T" in the middle of the model name it means it is the "standard" display. Ended up bricking the display and having to reflash it first lol. Now after all this effort it still didn't work right. Only after I did a full on reset of the panel via completely restoring it to stock that it took the newer 1.05 blueprints well. I did not find 1.06 blueprints but using that esp firmware had linked up here it eventually all worked with the 1.05 versions. It all seems rather clunky still though. I wish there was one "all encompassing" blueprint for toggling options on/off and reloading all these custom buttons and such so that each time an upgrade happens there is no need to go through 10 different automations and reupdate all the blueprints and reflash the whole ESP prior also. I am sure you have enough headache with these but if you get some time in the future that would be really cool to have.

aderusha, 2024-05-12:
Man I hear you and I've been begging for some way to handle Blueprint updates for years.  If you can, maybe [add your voice to the feature request I started here](https://community.home-assistant.io/t/allow-blueprint-upgrades/366939).

WebDesignerlol, 2024-05-12:
> Man I hear you and I've been begging for some way to handle Blueprint updates for years. If you can, maybe [add your voice to the feature request I started here](https://community.home-assistant.io/t/allow-blueprint-upgrades/366939).

just did. I also wanted to point out that I have somewhat used custom buttons on my HASPone as well, which isn't technically documented in the documentations but I stumbled upon by pure experimentation. so my panel looks like this for Page 8 (which is something that shouldn't be possible, as the slider on the second to bottom row is from another page and so on) 
[image]
Just thought maybe it would be a good idea to throw into documentation somewhere that technically this usage case is possible but may come with its own errors (like sometimes colors have a struggle loading properly, which i solved with more automations and some delays) I belive this may be due to memory issues on the ESP used in these.

omayhemo, 2024-05-19:
Using the latest bin and deleted and reinstalled all bps.  Also factory reset hasp. Now it’s working again. Thanks!🙏

aderusha, 2024-07-24:
Looks like 1.06 resolved this, closing issue.