HASwitchPlate/HASPone #248: Need to Reset HASPone

stuartdhii, 2024-08-21:
Screen boots to: WIFI Connected!
                             IP:192.168.50.95

                             Configure MQTT:
                             http://192.169.50.95

I cannot access the setup screen and need to reset the device.  I even tried static setting for Homeassistant at this address to try and bring up MQTT and it then rebooted with another assigned ip address for both the device and MQTT. 

I purchased the unit build from you a few months ago and am now trying to bring it up. Let me know what I can do to reset. 

Thanks,  Stuart

BenJewell, 2024-08-26:
Oh wow, I am having the exact same issue, and I also purchased mine a few months ago and just started setting it up! Spent an hour messing around with it and thought I was crazy. I wounder if there was a specific batch that ours were both part of that they flashed with bad firmware.

Pretty silly that they built this with no physical reset button. We both paid a premium buying it pre-assembled and now it looks like we are going to have to learn how to flash it anyway. :(

aderusha, 2024-08-26:
Hey there!  OK so one thing that might help right away would be to turn off your WiFi in your house, reset the HASPone, then let it go into setup mode when it cannot connect to WiFi.

Annoyingly, the next step would in fact be to take the device apart and run a factory reset if it's still not accessible.  Complete instructions can be found here: https://github.com/HASwitchPlate/HASPone/wiki/Flashing-HASPone-to-an-ESP8266

stuartdhii, 2024-08-26:
There is a physical reset button but it should say reboot as it does not
reset to factory setting.

austwhite, 2024-10-26:
@stuartdhii 
The button is a reboot, not a reset.  This only resets the the ESPdevice.  The previous instructions @aderusha  gave work.

BenJewell, 2024-11-18:
Thanks, I got it working after a couple tries. The integrated web server is really buggy and sometimes parts of the page don't load, including the required fields.

austwhite, 2025-04-22:
Just a note, remember th WeMOS D1 Mini module is a generic ESP8266 Mini module, not designed by the HaspOne project.   The "reset" button has always been a cold reboot button, not a factory reset button.  This is a design of the D1 Mini board so can't be attributed to the HASPONE project.

aderusha, 2026-03-28:
Closing this as I believe everything was addressed.