HASwitchPlate/HASPone #254: HTTP server doesn't fully load

nemith, 2025-03-10:
I have tried restarting and flashing over USB and after I setup WiFI i try to connect to the new IP address and only a partial HTML page is returned so I cannot configure the rest of my HASPone.

```
fw01% curl -v 192.168.12.159
*   Trying 192.168.12.159:80...
* Connected to 192.168.12.159 (192.168.12.159) port 80 (#0)
> GET / HTTP/1.1
> Host: 192.168.12.159
> User-Agent: curl/7.88.1
> Accept: */*
>
< HTTP/1.1 200 OK
< Content-Type: text/html
< Accept-Ranges: none
< Transfer-Encoding: chunked
< Connection: keep-alive
< Keep-Alive: timeout=2000
<
<!DOCTYPE html><html lang='en'><head><meta name='format-detection' content='telephone=no'><meta charset='UTF-8'><meta  name='viewport' content='width=device-width,initial-scale=1,user-scalable=no'/><title>HASPone </title><!DOCTYPE html><html lang='en'><head><meta name='format-detection' content='telephone=no'><meta charset='UTF-8'><meta  name='viewport' content='width=device-width,initial-scale=1,user-scalable=no'/><title>HASPone </title><script>function c(l){document.getElementById('s').value=l.getAttribute('data-ssid')||l.innerText||l.textContent;p = l.nextElementSibling.classList.contains('l');document.getElementById('p').disabled = !p;if(p)document.getElementById('p').focus();};function f() {var x = document.getElementById('p');x.type==='password'?x.type='text':x.type='password';}</script><style>.c,body{text-align:center;font-family:verdana}div,input,select{padding:5px;font-size:1em;margin:5px 0;box-sizing:border-box}input,button,select,.msg{border-radius:.3rem;width: 100%}input[type=radio],input[type=checkbox]{width:auto}button,input[type='button'],input[type='submit']{cursor:pointer;border:0;background-color:#1fa3ec;color:#fff;line-height:2.4rem;font-size:1.2rem;width:100%}input[type='file']{border:1px solid #1fa3ec}.wrap {text-align:left;display:inline-block;min-width:260px;max-width:500px}a{color:#000;font-weight:700;text-decoration:none}a:hover{color:#1fa3ec;text-decoration:underline}.q{height:16px;margin:0;padding:0 5px;text-align:right;min-width:38px;float:right}.q.q-0:after{background-position-x:0}.q.q-1:after{background-position-x:-16px}.q.q-2:after{background-position-x:-32px}.q.q-3:after{background-position-x:-48px}.q.q-4:after{background-position-x:-64px}.q.l:before{background-position-x:-80px;padding-right:5px}.ql .q{float:left}.q:after,.q:before{content:'';width:16px;height:16px;display:inline-block;background-repeat:no-repeat;background-position: 16px 0;background-image:url('data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAAGAAAAAQCAMAAADeZIrLAAAAJFBMVEX///8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAADHJj5lAAAAC3RSTlMAIjN3iJmqu8zd7vF8pzcAAABsSURBVHja7Y1BCsAwCASNSVo3/v+/BUEiXnIoXkoX5jAQMxTHzK9cVSnvDxwD8bFx8PhZ9q8FmghXBhqA1faxk92PsxvRc2CCCFdhQCbRkLoAQ3q/wWUBqG35ZxtVzW4Ed6LngPyBU2CobdIDQ5oPWI5nCUwAAAAASUVORK5CYII=');}@media (-webkit-min-device-pixel-ratio: 2),(min-resolution: 192dpi){.q:before,.q:after {background-image:url('data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAALwAAAAgCAMAAACfM+KhAAAALVBMVEX///8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAADAOrOgAAAADnRSTlMAESIzRGZ3iJmqu8zd7gKjCLQAAACmSURBVHgB7dDBCoMwEEXRmKlVY3L//3NLhyzqIqSUggy8uxnhCR5Mo8xLt+14aZ7wwgsvvPA/ofv9+44334UXXngvb6XsFhO/VoC2RsSv9J7x8BnYLW+AjT56ud/uePMdb7IP8Bsc/e7h8Cfk912ghsNXWPpDC4hvN+D1560A1QPORyh84VKLjjdvfPFm++i9EWq0348XXnjhhT+4dIbCW+WjZim9AKk4UZMnnCEuAAAAAElFTkSuQmCC');background-size: 95px 16px;}}.msg{padding:20px;margin:20px 0;border:1px solid #eee;border-left-width:5px;border-left-color:#777}.msg h4{margin-top:0;margin-bottom:5px}.msg.P{border-left-color:#1fa3ec}.msg.P h4{color:#1fa3ec}.msg.D{border-left-color:#dc3630}.ms* Leftovers after chunking:  292u bytes
* Connection #0 to host 192.168.12.159 left intact
g.D h4{color:#dc3630}.msg.S{border-left-color: #5cb85c}.msg.S h4{color: #5cb85c}dt{font-weight:bold}dd{margin:0;padding:0 0 0.5em 0;min-height:12px}td{vertical-align: top;}.h{display:none}button{transition: 0s opacity;transition-delay: 3s;transition-duration: 0s;cursor: pointer}button.D{background-color:#dc3630}button:active{opacity:50% !important;cursor:wait;transition-delay: 0s}body.invert,body.invert a,body.invert h1 {background-color:#060606;color:#fff;}body.invert .msg{color:#fff;background-color:#282828;border-top:1px solid #555;border-right:1px solid #555;border-bottom:1px solid #555;}body.invert .q[role=img]{-webkit-filter:invert(1);filter:invert(1);}:disabled {opacity: 0.5;}</style><style>button{background-color:#03A9F4;}body{width:60%;margin:auto;}input:invalid{border:1px solid red;}input[type=checkbox]{width:20px;}.wrap{text-align:left;display:inline-block;min-width:260px;max-width:1000px}</style></head><body class='{c}'><div class='wrap'><h1>%
```

The entire html body is missing.  I am out of ideas.

aderusha, 2025-03-12:
Has this worked previously?  One thing I've discovered is that, for whatever reason, my desktop PC will see this behavior but connecting to the device with my phone will load the whole page.

drderiv, 2025-05-19:
I had this same issue and while I have not had time to fully debug it, it appears to be happening because the variable `haspNode` has been corrupted.  In `webHandleRoot()`, the value of `haspNode` is inserted into the header, and as can be seen in the above `curl` output, it shows as blank when cast to a `String()` and inserted into the `<title>` tag.  Then, after an `<h1>` tag is sent, the corrupted `haspNode` variable is passed to `sendContent()`, and the closing `</h1>` tag is never received, presumably because the corruption causes an untimely exit from `webHandleRoot()`.  So while `String()` is able to gracefully handle the corruption, `sendContent()` is apparently not set up to do likewise.  How `haspNode` gets corrupted I am unsure.  I started with a clean install, connected to the HASPone's AP, and my Android phone immediately went into captive portal mode where I input only my wifi SSID and pw, which successfully got the HASPone onto my network (but without the proper mqtt parameters, which I did not input).  I don't recall seeing the other inputs but `haspNode` is pre-defined and should have been set to the default value, so how it gets corrupted needs to be chased down.

Fortunately with the help of @aderusha, I found a workaround.  Simply paste the following URL -- with parameters replaced for your setup -- into your browser and you'll be back on track.

http://_HASPone_ip_address_/saveConfig?wifiSSID=_wifi_ssid_&wifiPass=_wifi_password_&haspNode=plate01&groupName=plates&mqttServer=_mqtt_server_ip_address_&mqttPort=1883&mqttUser=_mqtt_username_&mqttPassword=_mqtt_password_