HASwitchPlate/HASPone #260: No text is displayed / nextion_return_data Invalid Variable name or attribute

nathanstitt, 2025-08-25:
I had a very old HASPone sitting around that I finally installed, my guess these issues relate to older hardware but I'm not sure.  Hopefully someone can point me toward additional debug steps or a firmware update I missed.  Thank you in advance for any assistance!

The device does not display any text and no blueprints other than core appear to work.

I've updated to the latest firmware 1.06 and applied the LCD update for LCD model `NX3224F024_011R`

I am able to switch between the pages using home assistant and it does change the layout, but no text appears on the bottom row as it should.  Adding blueprints appears to work, but they have no effect and do not display.  

Using telnet debug, I see the below errors displayed (I named the device `kitchen_switch`):
```
 'hasp/kitchen_switch/state/json' : '{"event":"nextion_return_data","return_code":"0x1a","return_code_description":"Invalid Variable name or attribute"}'
```

The webpage shows this debug info:

```
MQTT Status: Connected
MQTT ClientID: kitchen_switch-4c75251ad8b
HASPone FW Version: 1.06
LCD Model: NX3224F024_011R
LCD FW Version: 3
LCD Active Page: 1
LCD Serial Speed: 115200
CPU Frequency: 160MHz
Sketch Size: 672144 bytes
Free Sketch Space: 2469888 bytes
Heap Free: 17432
Heap Fragmentation: 11
ESP core version: 3.1.2
IP Address: 192.168.1.52
Signal Strength: -64
```

The screen appears like:

[image]

nathanstitt, 2025-08-25:
Oh and I forgot to mention that the firmware update url site haswitchplate.com seems to be down, I downloaded firmware from 
https://github.com/HASwitchPlate/HASPone/raw/b2c263eee772071df3abd3b464ad38737e848bfa/Arduino_Sketch/HASwitchPlate.ino.d1_mini.bin  and https://github.com/HASwitchPlate/HASPone/tree/main/Nextion_HMI and uploaded them to the device page

aderusha, 2025-11-22:
for whatever it's worth, the option "First-time HASP setup or reset" in the Core Functionality blueprint will re-create these devices for you in the event they wind up missing.

Details: https://github.com/HASwitchPlate/HASPone/wiki/Fix-common-HASPone-issues#re-run-core-functionality-first-time-hasp-setup-or-reset